procount/pinn #557: XBian is not installed probably with "ERROR 404: Not Found"

ssakatausa, 2021-09-09:
I tried to install XBian on RPi 3B using pinn. My attempt failed with the "OS(es) installed with errors" message. The debug log shows that pinn fails to retreive the XBian boot file, and it seems that the URL of the file shown in the log is incorrect. I would greatly fix the problem.

For your information I have attahced the photo of the log.

Thanks!!

[image]

Borjatgalan, 2022-02-20:
Today this error is still ocurring.

procount, 2022-02-21:
Sorry, I've not had time to look at this yet.

procount, 2022-02-21:
URL updated. Installed on Pi4/PiZero2 ok.

closed 2024-04-26 by procount