procount/pinn #558: How add Hyperpixel3 driver to buildroot

l0l0l0, 2021-09-09:
Hello, Ii am trying to add your kernel drivers in Buildroot, but i dont really know how to go about it. can you help me? or tell me how to start.
I want to compile it for a pi3 with this version
### [(https://www.furkantokac.com/rpi3-fast-boot-less-than-2-seconds/)](url)

Thank you and good day

procount, 2021-09-09:
My hyperpixel kernel drivers can be found in buildroot/linux as patch files. Look for one with Pimoroni or hyperpixel in the name. You may have to go back some commits to find the last version as I recently disabled them. Try to use the latest one I have.  I usually disable the patch files by adding .dis to the end of the filename, so remove that to enable them. After that, you need to do a `make clean` on buildroot and then rebuild buildroot after enabling the drivers in the linux kernel configuration.

The patch files are kernel version specific, so you may have to modify them, particularly the linux-xxx/drivers/input/touchscreen/Kconfig and Makefiles. The patch may not be able to apply those files simply because the contents  of the original files have changed, so you can just edit the patch files to match - they just include the hyperpixel Kconfig and Makefile files.

If the drivers don't build with your kernel, you may have to fix them. Let me know if you make any changes so I can include them in my versions. The original source is in the Cypress3 folder.

l0l0l0, 2021-09-09:
Great, thank you very much for your detailed explanation. i will try all this

l0l0l0, 2021-09-14:
Hello, 
So I managed to compile everything with your patch, but it still doesn't work?
I have the files pimhyp3.0/pimhyp4.0 in linux-xxx/drivers/input/touchscreen and I have activated them with make linux-menuconfig. I also have the pimhyp*-overlay.dts files.

There is something I don't understand, do I have to convert the .dts to dtbo files? And in your config file, there is this line:
dtoverlay = hyperpixel-gpio-backlight ... can I take the one from Pimoroni and put it in the overlay directory?

Thank you again for your help

procount, 2021-09-14:
I think the files should be pimhyp3 and pimhyp4 (maybe that is what you meant?)
The DTS files should be located in buildroot/output/build/linux-xxxx/arch/arm/boot/dts
They should be converted to dtb files as part of the build process. I think this can be done manually with `make dtbs`.

Yes, you can use the pimoroni gpio backlight overlay.

l0l0l0, 2021-09-14:
Thank you for your quick reply.
Yes these are the files I have, sorry.
I think my problem is there..i will convert them
Thanks again !

procount, 2021-09-21:
Did you get it working?

l0l0l0, 2021-09-21:
Not yet, but since I'm trying to do a minimal version of the kernel, I think it's more due to my config ... that's why I closed the topic.
But it's nice to ask :-D

l0l0l0, 2021-10-18:
Hello,
Finally no longer needed, I found a replacement screen (https://shpi.de/index.php?route=product/product&product_id=68).
Anyhow, thanks for your help !

lurch, 2021-10-18:
Ahhhh, always really annoying when you spend ages trying to debug a software problem, only to discover that it's a hardware problem :grimacing: You have my sympathies!