procount/pinn #559: Cannot boot Lakka_RPi4 on Raspberry Pi 4B

Jonja1, 2021-09-24:
I had been running PINN lite and Lakka 2 on my Raspberry Pi 3B+. Ran fine. I recently upgraded to a Pi 4B and tried to upgrade Lakka to the latest version available on the 3 hoping it'd work on the Pi 4 as advertised. It didn't work. 

On my Pi 4, I used PINN lite to replace Lakka2 with Lakka 4. This has been a complete failure. After a very long install where it is stuck at 100% (like an hour) it says it is complete with errors 
Seems see debug. When I try to boot Lakka I see a Raspberry Pi default screen that says nothing has been installed on that partition. It never boots into Lakka. 

On that same partition I can install other OSes and they work fine, such as Kali. 

This is a dual boot. My other OS is Raspian buster and it works perfectly. 
[image]

[image]

procount, 2021-09-24:
That's odd. 
For various reasons, I had PINN v3.3.4 handy so I used it to install Lakka_Pi4. It installed and ran fine.
So then I installed Lakka_PI2, then replaced it with Lakka_Pi4 to try and replicate your issue. It installed and ran fine.
The only issue I saw was that it paused for about a minute at the end of the installation (i.e. 100%, like you said, but not an hour) - probably because the downloaded size has not been updated in a while, but other than that it installed and ran fine.
DId you look at the logs? (ctrl-alt-f2, login with root/raspberry, then `less /tmp/debug`)

Jonja1, 2021-09-24:
I took pictures of the contents of the debug file
[image]
[image]
[image]
.  This is the 5th time it has failed.

Jonja1, 2021-09-24:
Just installed LibreElec on that partition as well, just for testing. So far it is only Lakka that won't install.

procount, 2021-09-26:
It looks like the tar file is not being written properly to the partition, although I've never seen the "Write to restore size failed" error message before. What is the size of the partitions you are trying to install to? I wonder if the new image is just too big...?

Jonja1, 2021-09-26:
It is a 64GB card partitioned into two 32 gb main partitions. Lakka 2 was written to this partition previously but wouldn’t boot the new Pi 4. Been trying to reinstall Lakka 4 on the exact same partition.

procount, 2021-09-26:
I really don't understand why it doesn't work for you, yet it installs ok for me.
You are letting PINN install lakka from the internet, rather than using a previously downloaded version?
I will repeat the installation with PINN 3.6 just to check that is also ok. Do you have another as card you can try it with?

procount, 2021-09-26:
I upgraded to PINN 3.6 and did a re-install of Lakka-pi4 and it was still fine.
It must be something with your SD card or your Rpi....

Jonja1, 2021-09-26:
Have been using PINN lite on this card for a couple of years with Raspian and Lakka just fine. I first upgraded Raspian to buster (success!) and now tried Lakka 2 to 4 since Lakka 2 wouldn't even boot on the new RPi4. This is where things went to crap with Lakka4. I have been able to install anything else to that same partition with no trouble. 

Lakka is the only other OS I want and the only one that doesn't play nice. Lakka team says PINN lite is not supported and I noticed they deprecated NOOBS and took down the tar balls for NOOBS from their latest and nightly builds sections. 

I have used PINN lite to upgrade to the latest version but didn't manually reinstall it.

procount, 2021-09-27:
I'm a bit short on ideas.
What I would do next is to prepare a USB stick for download and use PINN to download Lakka to it. Then use PINN to try and install Lakka from the USB stick.
If that fails with similar error messages, I would try to manually extract the tar file using bsdtar to the same partition, or even to an appropriately formatted partition on the USB stick, or to another SD card.
All PINN is doing here is crafting a command line to download the tar file and extract it to the partition. It seems to be bsdtar that is failing.

chenteparao, 2021-10-06:
> @chenteparao - this looks like the certificate error mentioned in #560. Version 3.6.2 should fix this. You can try the upgrade script, or wait until it is fully tested.

indeed, thank you very much man

closed 2021-12-28 by procount

mcarni, 2024-11-24:
> I'm a bit short on ideas. What I would do next is to prepare a USB stick for download and use PINN to download Lakka to it. Then use PINN to try and install Lakka from the USB stick. If that fails with similar error messages, I would try to manually extract the tar file using bsdtar to the same partition, or even to an appropriately formatted partition on the USB stick, or to another SD card. All PINN is doing here is crafting a command line to download the tar file and extract it to the partition. It seems to be bsdtar that is failing.
> 
> EDIT: a bit more googling seems to suggest my original hunch was correct and that you are running out of disk space when extracting (http://flummox-engineering.blogspot.com/2014/10/homebrew-write-to-restore-size-failed.html), which seems odd given the reported 32GB partitions you said you are using.
> 
> Please post the output of `fdisk -l`. You could also mount the appropriate partitions you are trying to write to and post the output of `df -h`

is there any update after this?
I am in the same situation of @Jonja1, i definitively can install other os'es (i already tried both retropie and liberelec) but i cannot install Lakka (neither 32 nor 64 bit)
i get the same message "installed with some errors" and the "...write to restore size failed"

would it make sense to reopen this issue?
is there any info i can provide to support?

thanks a lot

procount, 2024-11-24:
Which model of Pi are you using?
Which version of PINN?

mcarni, 2024-11-24:
sure, raspberrypi 4
PINN v3.9.2 built May 22 2024

procount, 2024-11-24:
Are you doing a fresh install, or replacing a ProjectSpace or other OS?

mcarni, 2024-11-24:
i used to have Raspberry PI OS + Libreelec
i tried to replace Libreelec with Lakka (tried both 32 and 64 bit), could not make it
I replaced Libreelec with Retropie and it worked (I used the same process, i wanted to check if i was doing something wrong)
then i tried a couple of times to replace Retropie with Lakka and it fails
(sometimes it takes a bit longer, not 1 hour as in the OP)

procount, 2024-11-24:
I've just tried a fressh install of Lakka_64 on a PI4 and it installed fine. Lakka_32 is being installed as I type.
I think the problem is that although Lakka has a 2 partition layout and could potentially replace a similar OS, the boot ("system") partition needs to be 2GB and the "storage" partition is a minimum of 512MB. So the unzipping of the tar file is failing due to a lack (lakk? :wink: ) of space. I'm afraid you'll have to do a fresh install, and then that should fix it.
PINN probably needs to do a few more checks that the partition sizes are large enough as well as being the correct number to cope with this odd case.

procount, 2024-11-24:
Lakka_32 has just finished installing successfully.

mcarni, 2024-11-24:
ok, what about some fdisk to resize the System partition to be >2GB

`NAME        MAJ:MIN RM  SIZE RO TYPE MOUNTPOINTS
mmcblk0     179:0    0 29.5G  0 disk 
├─mmcblk0p1 179:1    0  127M  0 part /mnt/RECOVERY
├─mmcblk0p2 179:2    0    1K  0 part 
├─mmcblk0p5 179:5    0   32M  0 part /mnt/SETTINGS
├─mmcblk0p6 179:6    0  256M  0 part /mnt/System
├─mmcblk0p7 179:7    0 14.6G  0 part /mnt/Storage
├─mmcblk0p8 179:8    0  510M  0 part /boot/firmware
└─mmcblk0p9 179:9    0   14G  0 part /`

obviously after a good backup ;-)

M

procount, 2024-11-24:
Yes, that should be fine. Shrink p7 and move it right. Then expand p6 to 2GB.
Fdisk should be fine, but I've found Gparted (maybe parted too) often changes the diskid, which updates all the PARTUUIDS, then nothing boots (that uses PARTUIDS), requiring the original diskID to be restored.

mcarni, 2024-11-24:
ok, perfect, give me a couple of days.. and i will try.. and let you know

mcarni, 2024-11-24:
> Yes, that should be fine. Shrink p7 and move it right. Then expand p6 to 2GB. Fdisk should be fine, but I've found Gparted (maybe parted too) often changes the diskid, which updates all the PARTUUIDS, then nothing boots (that uses PARTUIDS), requiring the original diskID to be restored.

sorry, i guess i failed.. i thought i used fdisk successfully but evidently i didn't
i now have raspberrypi os on partition 6 and 7
but PINN rightly is trying to boot 9 which i suspect it has also some fs issue
I am not worried about p9 as i wanted to replace Retropie with Lakka anyway 
but i was wondering if there is a way to point PINN to boot from p7
(i tried changing installed_os.json swapping 6/7 with 8/9 and vice versa but i didn't work)

i know i brought this on myself... this will teach me a lesson... i can redo the whole card.. 
but if you have any idea on how to fix this it would be super appreciated... 

thanks

procount, 2024-11-24:
As you have found, messing with partitions can be fraught with problems. It's probably best to backup your OSes (if necessary) and do a complete reinstall.

mcarni, 2024-11-26:
hi @procount , i have good news and bad news... 

good news is that i managed to fix the mess i made with partitions...
i took some sorting, but in the end i managed it
pinn works fine, raspberrypi os boots fine, i can replace the second os, i tried Retropie and Libreelec and all works fine

bad news is that Lakka 64 doesn't want to boot
it installs fine but when i try to boot it drops into some a debug shell complaining about 
'Error in mount_flash: mount_common: Could not mount LABEL=System'
as said, i can replace Lakka with other operating systems and they boot fine so I am guessing there is some issue with Lakka itself
PS. It seems also it is not 100% consistent as i reinstalled several times both Lakka 32 and 64 bit and it booted at least once the 32 bit version..

I hope this helps

in any case, thanks for the support, much appreciated
M

procount, 2024-11-26:
Let me first say that I didn't convert Lakka for PINN. I think it was first done for NOOBS. So I don't know much about it.
But given your comment `Error in mount_flash: mount_common: Could not mount LABEL=System` I think there is a clash between the OSes that you are trying to multi-boot.

OSes normally consist of more than 1 partition. The OS will need to reference all of its partitions. For example, the boot partition needs to know where the main root partition is and the root partition will need to know where all its partitions are. So the partitions need to be referenced. References can either be by device (e.g. /dev/mmcblk0p6) by partuuid (e.g. 12345678-06) by UUID (e.g. 12324-5678-xxxx etc) or by label (e.g. 'System'). When multi-booting with PINN, the OS is using different partitions than it expects, so PINN will adjust the references appropriately. The device and partuuid references are device and position dependent (so PINN has to adjust them accordingly) but UUIDs and labels are independent. However, they still need to be unique, so if PINN detects a partition already exists with the same name, it will rename one of the partitions (by adding an index number) to ensure they are unique. So you can imagine that if you have 2 OSes with the same partition labels, and they rely on the partition labels to reference the partitions, there could be a problem.

There are some flags that can be added to the PINN meta files of an OS to ensure its partition labels are not adjusted, but this will just prevent you installing an OS if the required partition labels already exist. (Ubuntu is an example of an OS that uses partition labels, and hence only 1 OS instance can be installed at a time). I don't see any such flags for Lakka nor LibreELEC. I think Lakka and LibreELEC are related in some way because they both seem to use the same partition labels,

So I think you can install LibreELEC or Lakka, but not both together on the same multi-boot drive. I can take a look at adding the meta flags, but this won't solve this situation which would need resolving by the OS maintainers, it will just prevent you installing both OSes.