procount/pinn #560: PINN cannot connect to some networks due to SSL error

fu-sen, 2021-10-01:
Immediately after booting PINN, I received a download error:
[image]
Here is isshoni.org, but I can't download it either. Therefore, it also occurs in some repositories.

isshoni.org and the download repository are fine:
https://isshoni.org/pinn/os/

So when I browsed the web I got this:
[image]

Some sites can be referenced by this web browser and some cannot.
It seems that at least sites using Let's Encrypt cannot be referenced. 
(For example, `github.com` using DigiCert can be referenced normally.)

I predict that it is caused by the old SSL intermediate certificate that makes up PINN.
I am in Japan this time October 1st. You'll probably get this when your location is in October.
I was able to use it on September 30th without any problems.

lurch, 2021-10-01:
Might need https://github.com/procount/pinn/tree/master/buildroot/package/ca-certificates to be updated?

fu-sen, 2021-10-01:
I can't get the PINN console, but Let ’s Encrypt has this information. Unfortunately PINN got this:
https://letsencrypt.org/docs/dst-root-ca-x3-expiration-september-2021/

Will it be updated with a package update? Debian seems to have a command for that:
https://manpages.debian.org/buster/ca-certificates/update-ca-certificates.8.en.html

df8oe, 2021-10-01:
Here the same. Also I cannot access the lists via wget on console. Same message: certificate has expired.

procount, 2021-10-01:
I need to look into this.
Please bear with me and I will try to release a new fixed version as soon as I can.

fu-sen, 2021-10-01:
> @fu-sen: you can access console by pressing ALT-F2 . Login is root/raspberry.

Thank you for that information. Now that I've got PINN root, I'm looking for a workaround.

lurch, 2021-10-01:
I _suspect_ that there'll be no "workaround" as PINN is a "static" (minimal) system and doesn't support package updates. I think the only fix will be for @procount to build and release a new version of PINN?

procount, 2021-10-01:
@lurch - yes I think you are right, but I will put a temporary workaround in place for the short term.

fu-sen, 2021-10-01:
The only permanent workaround is to wait for the PINN to update, but if you have a console with root privileges, you should be able to get a root certificate and work temporarily.

df8oe, 2021-10-01:
That is absolutely correct. I by myself jumed to a "direct-Arch-installation-procedure" without any helpers like PINN. But of course I wait for PINN update because of many newcomers do not have the skills to fix things on console ;)

fu-sen, 2021-10-01:
Well, related information:
https://community.letsencrypt.org/t/production-chain-changes/150739

And it seems that PINN is adopting the Debian package, so it's `ca-certificates`. 
https://packages.debian.org/en/stretch/all/ca-certificates/filelist

I hope this is a hint for the solution.

fu-sen, 2021-10-01:
The same problem has already been reported in other environments. Either of these could be a workaround:
https://stackoverflow.com/questions/69387175/git-for-windows-ssl-certificate-problem-certificate-has-expired

onyxbeam, 2021-10-02:
So how are we able to fix this? I tried mounting the filesystem and searching around to find DST_Root to comment out but the 3 paritions for PINN aren't familiar to me and don't match what I expected. I need a little more help here on specifically how to remove or disable DST_Root -- keeping in mind the filesystem claims it's read-only once booted.

df8oe, 2021-10-03:
We have to wait until the package is recompiled. The file has to be removed before compiling.

procount, 2021-10-03:
I have tried updating the ca-certificate package and manually removing the DST-ROOT certificate, but no difference.
I have still to find the new certificate and check it is included in ca-certificate and add it if it does not exist.

Other instructions from Let's Encrypt suggest that openssl needs upgrading from 1.02 to 1.1.0 or later. I have tried doing this, but the openssl package in buildroot changes significantly between these 2 versions, so it may mean a complete upgrade of buildroot (which is non-trivial).

Initially I thought this only affected the gentoo64 installations from Sakaki, so I was going to move them to my sourceforge repository for a quick fix, but then I realised it affects all of that as well.

The only workaround you can do yourself for the moment is to manually download an OS folder from sourceforge using an alternative browser and store it on a local USB stick in a /os folder for local installation (see https://github.com/procount/pinn/blob/master/README_PINN.md#local-images-on-usb-stick for details).

df8oe, 2021-10-03:
Have you tried to replace the X3 certificate by this certificate: https://crt.sh/?id=9314791 ? Of course you have to use the .crt format ...

narspt, 2021-10-06:
On an old browser I had to include Let’s Encrypt R3 intermediate cert, as just adding new ISRG Root X1 and removing DST Root CA X3 wasn't suffice, not sure why...

procount, 2021-10-06:
Thanks for this hint. I had already considered it, and it may be a necessary stop-gap.
Where I use `wget` this is quite easy to do, but in many cases I use QT's network classes which require a bit more work. I am investigating this in the background.

df8oe, 2021-10-06:
https://stackoverflow.com/questions/55194575/how-to-ignore-certificate-error-on-qml-webengineview

procount, 2021-10-06:
Looks like I have a workable solution!
I'll do a bit more testing and prepare a release so I can push up a new version ASAP (maybe later today or tomorrow)
Thanks to all for your suggestions.

df8oe, 2021-10-06:
I will test it.

procount, 2021-10-06:
Here is the upgrade script
https://sourceforge.net/projects/pinn/files/archive/upgrade362.sh/download
On a PINN v3.6 SD card (that has the certificate error), copy this script to the PINN partition.
Boot into PINN. 
Either enter the recovery shell (press ctrl-alt-F2) or connect to the shell over ssh, and login with root/raspberry.
Then type the following commands:
```
cd /mnt
./upgrade362.sh
```
The v3.6.2 version will be downloaded, installed and then PINN will be rebooted.

fu-sen, 2021-10-07:
This comment was increasing at midnight in Japan.
I forgot this report: Initially I just got the Japanese version of the Raspberry Pi 400 and was preparing the environment with PINN.
I installed PINN and scripts on the SD card with RPi400 and tried it.

[image]
[image]
[image]
[image]

It's working! My RPi400 and PINN got the expected behavior.

procount, 2021-10-07:
Thanks for testing. I think Raspberry Pi OS would install anyway, but I guess the rest of the errors you previously reported have gone anyway now.