procount/pinn #561: OS Boot Selector is bypassed on bootup

waltwright, 2021-10-02:
OS Boot Selector is bypassed on bootup. Work-around is to hit shift and Exit to OS Boot Selector.

I have at least 2 OS systems installed and configured. I have a data partition configured, 4 project spaces, 9 OS systems installed.
I experimented with vncshare, forcetrigger, remotetimeout successfully.
The system stopped entering OS Boot Selector during that experimentation time period, so I used the Maintenance menu to remove those configurations. What remains in the recovery.cmdline is below; however, the OS Boot Selector still doesn't work.

forceupdatepinn select="allinstalled" no_default_source disablesdimages alt_image_source=http://pinn...... no_group quiet ramdisk_size=32768 root=/dev/ram0 init=/init vt.cur_default=1 elevator=deadline loglevel=2 sdhci.debug_quirks2=4

procount, 2021-10-02:
When you arrive at the boot selector menu, are any of the checkboxes to the left of any of the OS selections checked? If so, remove the check.

waltwright, 2021-10-03:
Thanks, I appreciate the help.
Indeed, I removed the checkmark from an OS at the boot selector menu and now it boots through to the OS Boot Selector properly.

closed 2021-10-03 by waltwright