procount/pinn #568: Update to match noobs 3.7?

m-roberts, 2021-10-13:
NOOBS 3.7 release is [here](https://github.com/raspberrypi/noobs/commit/5de0ec896cefff9f095d3243ee41bcce392ce0ef), which contains new firmware - including, I believe, support for 8GB Pi 4?

procount, 2021-10-13:
https://sourceforge.net/projects/pinn/files/

m-roberts, 2021-11-03:
> Just a small note. I've been using PINN on a Pi 4 8gb for like one and a half years now.
> 
> I had some issues that caused the Pi not to boot when I had a Monitor connected which was turned off. But that was fixed in one of the updates.

Thanks for this. It seems I was way off. At any rate, updating to latest firmware is usually a good move!

procount, 2021-12-03:
PINN is now up to v3.7.3 so I think we can close this now.

closed 2021-12-03 by procount