procount/pinn #569: Install so

progamer562, 2021-10-16:
Is there a way to install another os instead over writing all the os installed

procount, 2021-10-16:
Yes. It is documented in the manual.
Have a look in particular at:

https://github.com/procount/pinn/blob/master/README_PINN.md#project-spaces

https://github.com/procount/pinn/blob/master/README_PINN.md#reinstall-individual-oses

https://github.com/procount/pinn/blob/master/README_PINN.md#replace-individual-oses

progamer562, 2021-10-16:
I don’t think so what I mean that berryboot allows you to install an os without Berryboot removing anything on the sd card 

or the links you send are right

but what are the down sides tho

procount, 2021-10-16:
PINN is based on NOOBS and it works in a different way to Berryboot.
The idea with NOOBS/PINN is that it allocates all the drive space to the selected OSes, so there is never any space left to install something else later - you always need to replace an existing OS.

progamer562, 2021-10-17:
Ohhh ok what happens if I use project space will it let me not reinstall any os?

progamer562, 2021-10-17:
I’m confused 

How do I config projects space tho
I’m new to pinn

procount, 2021-10-17:
No I don't have any videos. Maybe I should become a Youtuber!

You don't really config a project space. Depending on how big your drive is, just install 1 or 2 projectspaces, along with any OSes that you want to a fresh drive.
Later, when you want to install another OS, just select the OS you want, then select a free installed ProjectSpace from the maintenance menu and select Replace. Choose the new OS from the drop down list and replace the projectspace with your new OS. Done.

progamer562, 2021-10-17:
You should be a YouTuber 
If you did it will make it easy to use them

lurch, 2021-10-18:
IMHO the easiest way to get to learn how to use NOOBS / PINN if you aren't sure what you're doing (after reading the documentation), is to take a spare SD card (that you're happy to completely overwrite) and just spend an afternoon experimenting / playing with it, to get a feel for how the different features work. :slightly_smiling_face:

progamer562, 2021-10-18:
dang ok i wish there was videos to know of but ok i will use a different sd card and play with it

closed 2021-10-18 by progamer562