procount/pinn #574: Missing Instruction: How to uninstall an OS

Tabaluga, 2021-11-01:
Dear @procount ,

please include instructions in README.MD on how to uninstall an installed operating system.

best regards

procount, 2021-11-01:
You can't. You can only replace it with another. 
It can be hidden from the menu if you make it non-bootable. Is that sufficient?

The idea with NOOBS was to allocate all drive space at initialisation. It was not designed to remove OSes and add more later on. PINN has improved on that with the ability to reinstall or replace individual OSes, or to reserve space with Project Spaces which can later be replaced by another OS.

closed 2021-11-01 by procount

Tabaluga, 2021-11-01:
Thanks for your explanation. Now I understood this behavior and design.. Indeed I've searched already for some clues within the NOOBS documentation but found .. as you imagine .. nothing at all about this. Perhaps add this into the getting started guide / FAQ. best regards

lurch, 2021-11-01:
@procount Is it possible to "replace" an installed OS with a ProjectSpace? If so, would that count as "uninstalling" an OS? :wink:

procount, 2021-11-01:
Wow! Why did I never think of that?
Yes, for OSes with a "standard" partition layout