procount/pinn #575: Missing Information: Credentials for every OS

Tabaluga, 2021-11-01:
Dear @procount ,

please include the credentials in the README.MD for each supported operating system without having to research them online. I'm using PINN offline.

Maybe it would be sufficient to display the credentials on a slide during the installation.

best regards

procount, 2021-11-01:
By "credentials" do you mean the login username/password? It is already included.

- Go to the Maintenance menu
- highlight your OS 
- select Password
- check Show Password
- click Use Default
- Use cancel to exit

closed 2021-11-01 by procount

Tabaluga, 2021-11-01:
Thanks for your explanation. This helped a lot. Perhaps add this into the getting started guide / FAQ. best regards