procount/pinn #581: Missing Information / feature: Mac Address

Tabaluga, 2021-11-01:
Dear @procount ,

please add an option to show the mac address of all devices within PINN.

best regards

Tabaluga, 2021-11-01:
I want to know the mac address to configure my firewall and router. Since I'm blocking everything it was not possible for me to locate the mac address within PINN. best regards

procount, 2021-11-01:
Ah, you want to know the MAC addresses used by each OS that PINN has installed? (Not other DEVICES on the network).
Typically, all OSes will use the same MAC address as this is governed by Ethernet chip or wifi driver.
Open the recovery shell (ctrl-alt-f2, root/raspberry) and just type `ipconfig` which will give you a list of interfaces and their MAC addresses (HWaddr). This will only work with interfaces that PINN supports. Otherwise, you can do the same command in one of the OSes you have installed.

closed 2021-11-01 by procount