procount/pinn #583: Idea: Move the Changelog

Tabaluga, 2021-11-01:
Dear @procount ,

please move the Changelog into an own file, or perhaps you can use something like the [release notes](https://github.com/atom/atom/releases) of the atom editor.

They even included (connected) these information upon their [main website](https://atom.io/releases).

best regards

procount, 2021-11-01:
Like this file: [config.md](https://github.com/procount/pinn/blob/master/changes.md) maybe?

Tabaluga, 2021-11-01:
Yes. This looks great. perhaps you could merge both files. Best regards

procount, 2021-11-01:
Both files? What is the other file you mean?
There is already a history.md file, a changes.md file, a readme.md file on sourceforge and the first post on my sticky forum post contains a complete log too. I don't think I need any more change logs.

closed 2021-11-01 by procount

Tabaluga, 2021-11-01:
Thanks for your explanation and please excuse my missing knowledge.
To be honest I'm browsing from my phone, the user-agent is set to mobile and the code section is not expanded at all. 
I read through the readme.md, found the change-history and thought that it might be a good idea to move this into an new independent file.
Since the time of reading I didn't had the knowledge of the other changes.md and history.md files.
best regards

procount, 2021-11-01:
No worries :smile:

lurch, 2021-11-05:
Perhaps it's worth removing the "Change History" part of README.md, and just linking to one of the other files instead? :shrug: