procount/pinn #585: Missing Instruction: How to setup, config and use ProjectSpace(s)

Tabaluga, 2021-11-01:
Dear @procount ,

please add additional information about the ProjectSpaces.

> ProjectSpaces are empty OSes that can be used to reserve space for a future OS to be installed into.

I want to install three OSes:

1. libreelec
2. alpine (later)
3. retropie (later)

How many ProjectSpaces to I need to create?

How to install the OSes into the ProjectSpace?

best regards

procount, 2021-11-01:
Install LIbreElec + 2 projectSpaces
Later, replace a ProjectSpace with your required OS.

Tabaluga, 2021-11-01:
Thanks for your explanation. This helped me a lot. Perhaps add this into the getting started guide / FAQ. best regards

procount, 2021-11-01:
https://github.com/procount/pinn/blob/master/README_PINN.md#project-spaces ?

Tabaluga, 2021-11-01:
Thanks for your explanation. I've read this already but it was unclear to me when do I need to install ProjectSpaces. best regards

procount, 2021-11-28:
documentation improved

closed 2021-11-28 by procount