procount/pinn #589: add suport to pi zero w 2

zaballa, 2021-11-13:
it would be wonderfull to have support for this model, recalbox an retropie for pi zero doesnt work

lurch, 2021-11-14:
FYI there's a big difference between PINN itself (i.e. the installer-GUI) working on the Pi Zero 2 W, and each of the OSes installed by PINN (e.g. recalbox or retropie) supporting the Pi Zero 2 W. The latter is something that each individual OS needs to fix themselves.
(Although as the Pi Zero 2 W is so similar to the Pi 3B from a software PoV, I think each OS "just" needs updated firmware and a new DTB?)

procount, 2022-03-27:
PINN is already compliant. The other OSes will be updated in PINN as their maintainers update them.

closed 2022-03-27 by procount