procount/pinn #590: How to remove black borders

lollysan, 2021-11-16:
Hi, I've just installed PINN Lite on a 128GB Microsd card on my Rpi4, but I noticed black borders around the image on the main menu. I have a 42" 1080p LED TV. How can I remove them? Thanks.

procount, 2021-11-16:
PINN has a config.txt file, just like Raspberry PI OS.
You can edit it to configure the Pi for your monitor.
You might want to set `disable_overscan=1`
See https://www.raspberrypi.com/documentation/computers/config_txt.html#generic-display-options

lollysan, 2021-11-17:
I've tried editing the config.txt file from the recovery console, but I
found out that all the config files are in the /mnt folder and they are
read-only. Maybe I did something wrong? Thanks.

procount, 2021-11-17:
There are several ways to edit this file.
1. If you put the SD card in a Windows PC, you can edit the file directly on the RECOVERY partition
2. In the recovery shell, you first have to mount the partition read-write
```
mount -o remount,rw /mnt
cd /mnt
nano config.txt
sync
rebootp
```
3. Use the GUI in PINN
```
Go to the Maintenance Menu
Highlight the PINN entry (no need to tick the box, just make sure it is highlighted)
Select the Edit Config button
This brings up an editor to allow the config.txt and recovery.cmdline files to be edited for the selected OS.
```

lollysan, 2021-11-18:
I managed to edit the configuration by using the GUI in PINN, thanks for help.

closed 2021-11-18 by lollysan