procount/pinn #591: CEC not working on Panasonic TV

Nicko98, 2021-11-16:
Hi,
at first: thanks for the great work!

I ran into a problem regarding CEC.
I‘m using a Raspberry Pi 4 1GB with PINN to Dualboot RetroPie and LibreElec together with a Panasonic TV.
CEC works well on LibreElec, but doesn‘t work completely correct on PINN.
When I initially boot the Pi CEC doesn‘t work on PINN, but after booting into LibreElec and rebooting into PINN it works (and i can reboot as much as i want as long as the TV doesn’t get shut down). When I shutdown and restart the TV only, it stops working. I assume it’s an issue with the initialization of the CEC conection. Any idea for a fix? Maybe I can help debugging this… (but maybe i can’t do so this week)

floot, 2022-02-12:
Hi!
I have the exact same symptoms as you with my Panasonic TV. But sometimes PINN catches the HDMI-CEC commands at power-on, and sometimes not (forcing me to let my default OS boot, then reboot it so that PINN shows up again and finally works).
I just can't explain why sometimes it works at power-on, and sometimes not.

procount, 2022-02-14:
At least Panasonic seems better than LG! :smile: It does sound like it is not initialising properly. Maybe it needs a different command sequence 🤷‍♂️
See also #164. Maybe you could test Berryboot to see if CEC works better with that? Then I can try and implement any differences.