procount/pinn #593: Pimiga request

BenMtl, 2021-12-05:
Has any one tried or thought about making a Pimagi image for Pinn?? I think it would be great !

djchewmacca, 2021-12-13:
I have already done it.  You will still need to provide your own kickstart rom and put it in the KICK folder.

https://www.dropbox.com/s/9clb1o24nwt9k06/Pimiga2.0_NOOBS.torrent?dl=0

djchewmacca, 2021-12-13:
BTW. I've also done Supreme Ultra V1 which based off RetroPie.  Pretty good build IMO...

https://www.dropbox.com/s/1s826i6upjglpyi/SupremeUltraV1_NOOBS.torrent?dl=0

jgbrwn, 2022-02-21:
> I have recently done Pimiga 2.0, you will still need to provide your own kickstart rom and put it in the KICK folder.
> 
> Pimiga 2.0 torrent: https://www.dropbox.com/s/9clb1o24nwt9k06/Pimiga2.0_NOOBS.torrent?dl=0
> 
> PimigaMF 1.5 torrent: https://www.dropbox.com/s/mf2njdbtbwj3kx2/PimigaMF1.5_NOOBS.torrent?dl=0
> 
> All I ask is to keep seeding once you've downloaded it as I have limited upload bandwidth.

@djchewmacca 
I'm trying to DL the NOOBS PiMiga 2 torrent but it's stalled...
Any chance you could reseed?
Thanks!