procount/pinn #595: Cz translation

JanSykor, 2021-12-06:
How can I help with translating to czech language?
Where I can find some tutorial?
Thanks
Jan

procount, 2021-12-06:
Hi Jan,
Thank you for offering to translate - it is much appreciated.
It's not an area I am very familiar with but here is what I do know:

PINN is developed under Linux (Ubuntu 14.04 or 20.04) using the QT4.8 framework.
Translations are done using the QtLinguist tool.

See https://github.com/procount/pinn/blob/master/README_PINN.md#addingupdating-translations which includes links to the official qt translation tutorials.

In order to perform a pull-request with your updated translation, you should clone the PINN git repository.
Use https://github.com/procount/pinn/blob/master/recovery/translation_en.ts as the original English file and translate it to translation_cz.ts (I assume cz is the correct Czech translation code, but please check).

JanSykor, 2024-01-07:
Hello @procount 
Sorry for the big Delay, but I am not familiar with Github pull requests, Ubuntu and QtLinguist tool so I did all translation in notepad (I hope without mistakes) and related file I am sending in zip file. If you can add it to next version please?

I will check it then and make some adjustment if will be necessary. I am expecting also some changes related to Pi5.
Thanks for your big job
Jan
[Pinn_Cz_Translation.zip](https://github.com/procount/pinn/files/13855289/Pinn_Cz_Translation.zip)

procount, 2024-01-08:
Thank you for this.
Can I ask which version of PINN this relates to? (just so I can integrate it correctly. i.e. does it originate from a version in Dec 2021, or a more recent version?)

JanSykor, 2024-01-08:
I started in Dec 21, but version which I sent is  from July 23 (3.8.7) - last public version on Github
Thank you 
Jan

JanSykor, 2024-01-08:
Dear @procount 
Now I am checking other translation in pull requests. And I realized, that I have mistakes in my translation. I put my translation instead of "untranslated" and it must on another place
So I will send you new version of my translation.
When you plan building of next version?
Thanks for understanding
Jan

procount, 2024-01-08:
I have already fixed your syntax errors and built a new beta version.
I will post it soon so you can check your translations.

JanSykor, 2024-01-08:
Thank you So much
The repair was possible automatic or it was necessary to do by hand?
Jan

procount, 2024-01-08:
I automated the changes.

procount, 2024-01-09:
You can download it from here -> https://sourceforge.net/projects/pinn/files/archive/pinn-387c.zip/download
Please let me know if you need any changes.

JanSykor, 2024-01-10:
Hello @procount 
Already tested on RPi 3B+ And system not booted. Please see picture.
New card, new installation.
If you need some other info, please let me know.
Thanks
Jan
[image]

JanSykor, 2024-01-13:
I am already tested your new Pi5 beta image and it is working good on Pí 3B+. Can you add the czech language to the next version please if on your testing version Is some kernel problem?
Thanks
Jan

procount, 2024-01-13:
Sorry, I only just saw your last 2 messages (I swear github is not sending me all my notifications!)
Hmm, I don;t know why you are getting a kernel panic - it worked ok for me. I'll check it.

procount, 2024-01-13:
I have uploaded a new copy of pinn-387c.zip - this one should be ok.
I noticed some text was still in English, so you may need to make some additions.
Also the name of the Czech language did not show up in the drop-down box. Can you provide a translation for that please?

procount, 2024-01-13:
Here is your modified translation file that you can update
[translation_cz.zip](https://github.com/procount/pinn/files/13929829/translation_cz.zip)

JanSykor, 2024-01-14:
Unfortunately I can not find in language file row for name of language
I found some handling with name of languages in languagedialog.cpp, in line 82 
but unfortunately I dont know how it works 
Correct name for Czech language is "Čeština"
Modified translation I will send later.
Thanks for your help
Jan

JanSykor, 2024-01-14:
I also found strange behaviour when you switching languages. Switching between UK And US english Is Ok, but when you will switch to another languages instead RPi model and memory volume Is show only  translation for model and memory. Repair Is possible only by switching to english and reboot.
This Is same in version 387 and 387c.
Jan

procount, 2024-01-14:
> Correct name for Czech language is "Čeština"

Thanks, that's all I need, I'll do the rest.

> when you will switch to another languages instead RPi model and memory volume Is show only translation for model and memory.

Thanks, no-one reported this yet. I'll take a look.

JanSykor, 2024-01-14:
Hello
How can I open Options dialog?
There is plenty of translation, which I dont known how to translate.
Thanks 
Jan

procount, 2024-01-14:
From the recovery menu, press 'M' twice to get to the Maintenance Menu.
Highlight the top PINN entry.
Select Edit configuration, then click the Edit Options button (Bottom left)

JanSykor, 2024-01-20:
Hello @procount 
I am sending revised version of czech translation. Sorry for the delay.
Finaly I decided some words (Virtual keyboard, configuration menu etc.) not translate, because there is translated help tip text and in wiki and github issues you use english name of parameters so for now I keep it not translated. There is also problem, that sometime in translated word is missing the letter for shortcut (after &amp;).
Also I can not find in the file name of tabs "Testing", "Sailing" and "Legacy"
Starting words " For Recovery mode, hold SHIFT ...." shows in testing version every time in english, but it is already translated in the file.
I am expecting next revision, of translation when you push new version for Pi5 (or before when you have it ready and if you send me new translation file - up to you)

Many thanks for your amazing job

Jan

[translation_cz.zip](https://github.com/procount/pinn/files/13999232/translation_cz.zip)

procount, 2024-01-21:
There are several issues with the translation process that could do with some improvement. I'm not sure how to deal with some of them e.g. the shortcut indication key using the '&'. I'm particularly interested in words/phrases that are not being translated, as it means I have missed them out in the code and they are generally easy to fix. Some come from external data files, though so they are not so easy to anticipate, like the group categories, unless I standardize on them.
Thanks for the update.

procount, 2024-01-21:
The ampersand character '&' displayed as `&amp;` is used to precede the shortcut character that you can use to select that button using the keyboard. In English they are chosen so that each letter is unique within a particular context (e.g. dialog window). If they are changed in a translation, then:
1) The translator must ensure they are still unique within that context.
2) The code would need to be adapted to use the new shortcut key.
In some dialogs, this will be done automatically, but in others I would have to do this myself and currently I'm not sure how to do this.
All I can suggest at the moment is to keep the shortcut key the same using perhaps the following example:
```
        <source>&amp;Edit Options</source>
        <translation>Změna nastavení(&amp;E)</translation>
```

procount, 2024-01-21:
Not quite.
To be honest I am not an expert in translations, but let me give you some more details.
The translation_cz.ts file becomes part of PINN's source code and it therefore has more effect than what you see on the screen.
If you put an `&amp;` before a letter in the name of a button or menu item etc, then it makes the following letter the shortcut letter for that option. So in the case of `<source>&amp;Edit Options</source>`, it makes `E` the shortcut letter to select that option.
If you change it to `<translation>noobs nastavení (c)</translation>` without the `&amp;` then there is no shortcut key.
If you change it to `<translation>noobs nastavení (&amp;c)</translation>` then you have changed the shortcut key to `c`.

Normally, the program will take care of using this new key `c` as the shortcut key, but this causes other problems.

1. The translator must make sure the same shortcut key is not used more than once on the same menu or dialog box.
2. Any documentation that refers to the original shortcut key `e` will now be wrong for that language. It would require translations of the documentation for every language and will require the correct shortcut keys documented to be changed to be correct for that language. That is an even bigger task.
3. In PINN there are additional mapping files for TV remotes using CEC and joysticks that can map certain buttons to the shortcut keys. If you change the shortcut key, these files will no longer work when your language is selected.

So I would propose that you keep the shortcut keys the same.
If your translation includes the same letter, then you could highlight the same letter. e.g.
`<translation>noobs nastav&amp;ení</translation>` 
even though this may not make much sense in your language.
If the translation does not include the same letter, then I propose you add it in brackets at the end, e.g.
`<translation>noobs nastavení (&amp;E)</translation>`
This will keep `E` as the shortcut key and cause it to be underlined to show the user.

I know this is not ideal, but at least it won't break the functionality.
There are other issues with translations, like inserting parameters that result in them being in the wrong place in the sentence for certain languages, or incorrect use of plurals etc. Unfortunately, these are not high on my priority list, but I hope to fix them someday,

And by the way, I found that the correct ISO code for Czech is `cs` so we need to use this in e.g. `translation_cs.ts` and `cs.png` etc. Using this code, the correct language name of `Čeština` will be used automatically.

JanSykor, 2024-01-22:
I donť know that for shortcut is this  `&amp;`    necessary (I think that it is hardcoded), because here is also not added in the source, and it is working in the program
```
        <location filename="mainwindow.ui" line="433"/>
        <source>More (m)</source>
        <translation>Více (m)</translation>
```
This is correct source, so I think C letter is correct:
```
<source>noobs&amp;config</source>
<translation>noobs nastavení (&amp;C)</translation>
```
Sorry for my mistake with ISO code - in the attachment you can find all files repaired

[translation_cs.zip](https://github.com/procount/pinn/files/14005983/translation_cs.zip)

Thanks 
Jan

closed 2024-01-29 by procount