procount/pinn #596: Updated Volumio to 3.165 Ethernet not working

alfieoriginal, 2021-12-08:
Hi, after updating to new Volumio 3 (3.165) wired network is not working.
After trying to change network config:

"Error: Command failed: /usr/bin/sudo /bin/ip addr flush dev eth0 && /usr/bin/sudo /sbin/ifconfig eth0 down && /usr/bin/sudo /sbin/ifconfig eth0 up Device "eth0" does not exist"

RPI3
Pinn 3.7.4
Wireless is working.
Can anyone reproduce it?

Thank you in advance

procount, 2021-12-24:
@alfieoriginal - Volumio was converted for PINN by @matthuisman, but it's really just a change in format - the content does not differ from the original image apart from references to its new partition location.
The error mentioned above does not look like it comes from the relocation of Volumio to a new partition, but it could be a possibility (i.e. maybe the upgrade is not aware of its new location).
Have you tried installing the updated Volumio to a fresh SD card? Does that exhibit the same problem?
You could also try any official Volumio forum e.g. https://community.volumio.org/

alfieoriginal, 2022-01-02:
Hello @procount  I've installed the new Volumio to a fresh SD card and does not exhibit the same problem. I can see and config the wired network. The issue appears when upgrading Volumio on Pinn. I can't find any other related post at Volumio forum.

Tex-1982, 2022-02-06:
Hello I have the same issue. Did you manage to solve it? If I reflash a new sd card and start from zero with ethernet connection every thing ok.... but we the old one with the wifi already setted I am not able to start ethernet connection

matthuisman, 2022-02-08:
taking a look at this now. So, this is new volumio3 - i assume they have changed the init script etc.
I won't be looking at fixing 2>3 upgrade path - but i will see if I can get a fresh 3 working at least

procount, 2022-02-08:
Thanks Matt,
I've read of users having issues upgrading Volumio from v2 to v3 without PINN, so it's probably more of a kernel related issue, like upgrading from Bullseye to Buster.

alfieoriginal, 2022-02-19:
Hi, reinstalling Volumio now solves the issue on RPI3.
Thanks for all @matthuisman and @procount

closed 2022-02-19 by alfieoriginal

paulwratt, 2022-02-21:
> Thanks Matt, I've read of users having issues upgrading Volumio from v2 to v3 without PINN, so it's probably more of a kernel related issue, like upgrading from Bullseye to Buster.

Some of these issues are related to RPi4 Revision # (according to one thread I read on RPi Forum).