procount/pinn #597: Problems with XBox 360 Joystick (Wireless) mapping

MopheusDG, 2021-12-10:
Hi everyone, I'm using a 360 XBox Wireless controller and most of the buttons do not work (No PAD moving, A button doesn't do anything, only X acts like ESC).
I would like to know if this is a mapping problem (Weird as XBox controllers seem to be supported natively), or is because of the Wireless version of the device ?

Any help will be appreciated. Thanks.

procount, 2021-12-10:
Can you post the debug log? (see troubleshooting in the wiki)

MopheusDG, 2021-12-11:
I'll check later today. So far my movements are:

Joy B (Red) => Moves DOWN
Joy LT (Left Trigger) => Moves UP
Joy X (Blue) => ESC
Joy Analogs => Go crazy with the mouse, movement is all over the place.

No button so select anything or to move to the top options at all. Tried all buttons, including analog sticks buttons.

Will add more later. Thanks.

MopheusDG, 2021-12-11:
I did some log review but I cannot see any error, seems to be some mapping problem maybe ? The device us being recognized fina as Xbox 360 Wireless Receiver and no errors are being reported.... Not sure how to copy the full log here.

procount, 2021-12-11:
Did you read Step 5 of https://github.com/procount/pinn/wiki/Troubleshooting?
Once you have your debug.txt file, create a new comment in this thread. At the bottom of the edit box, just above the green comment button, there should be a grey area that says "Attach files by selecting or pasting them here". Just drag and drop the file into that box, or click it to select the file from your PC.

If you try it on a mobile device, you may have to put the device in landscape mode for the box to appear.

At the top of the file it should say what the name of your joystick is. Then it will try to load a matching map file. If your device is a "Xbox 360 Wireless Receiver" I don't think that exists so it will load the standard joy_keys.json file. In which case you need to create the file it is looking for. You could start by copying one of the existing 360 map files to the correct file name.

See also https://github.com/procount/pinn/blob/master/joy_keys.md

MopheusDG, 2021-12-12:
Thanks @procount , finally make it work by manually copying the XBOX_360.json to joy_keys.json under the /mnt folder using the Recovery Shell and works perfectly fine. Seems the wireless controller is not being assign correctly, Will it help to upload the debug for future improvemnts ?

Thnks.

procount, 2021-12-12:
It's not the ideal way to do it but it will work.
Yes, if you can upload the debug log I'll tell you how to rename it properly and it will help others too.

MopheusDG, 2021-12-12:
/mnt/Xbox_360_Wireless_Receiver_8A_15B.json
[debug.txt](https://github.com/procount/pinn/files/7700034/debug.txt)

procount, 2021-12-12:
At the end of debug.txt it tells you what file it is looking for. As it does not exist it loads the default joy_keys.json.
So you should copy the XBOX_360.json file to /mnt/Xbox_360_Wireless_Receiver_8A_15B.json.

MopheusDG, 2021-12-13:
Works perfect now. Thanks for support @procount .

closed 2021-12-13 by MopheusDG