procount/pinn #598: LineageOS (KonstaKANG) 19.0 can't boot out of TWRP

masscream, 2021-12-13:
I've installed recently two OSes, LOS 19.0 and rpios. After a using the LOS for a while I wanted to flash Magisk so I rebooted to TWRP and my trouble began. There is probably an issue with mounting the system partition and therefore I can't see any content of the User storage. What is worse that I can't now boot back into the system. Any attempt leads to booting into TWRP again. Tried solving the root partition mount issue, however, not successfully so far. When I open "Mount" section, got there - Boot, System, Vendor, Data, USB, however only Boot partition can be marked. Here is some info

```
rpi4:/ # mount -a
mount: '/dev/block/mmcblk0p2' ->'/system_root': Invalid argument
1|rpi4:/ # cat /etc/fstab
/dev/block/mmcblk0p1 /boot vfat rw 0 0
/dev/block/mmcblk0p2 /system_root ext4 ro 0 0
/vendor ext4 ro 0 0
/data ext4 rw 0 0
/usb auto rw 0 0
rpi4:/ # mount -t ext4 /dev/block/mmcblk0p2 /system_root
mount: '/dev/block/mmcblk0p2' ->'system_root': Invalid argument
```
I guess it has to do something with the ext4 fs, but it should be supported. Otherwise I don't know.
OS version: LineageOS (KonstaKANG) 19.0 20211104 with TWRP 3.5.2_11-0

procount, 2021-12-14:
Yes, I confirm this is an issue. Something has changed between v18 and v19.
I was informed there may be a point at which LineageOS may not be multi-booted, due to the direction in which Android was heading. hopefully this is not it, but I will investigate to see what can be done.

procount, 2021-12-14:
Looks like this version of Android uses more advanced FS features that PINN currently does not support.  Looking into it.....

masscream, 2021-12-14:
In the meantime, guys on XDA helped me with a tempfix. I'm a little new for rpi stuff so it took me a while and a hint before I found out that the mounted stuff on TWRP (boot part) is actually the Pinn OS boot partition.... and the mmcblk0p2 could not be mounted because it's not the android root partition but extended p. for a number of logical partitions for two systems actually. Typing in the right devices' location (in my case mmcblk0p8 & 9, ...) into fstab and running mount -a mounts the right partitions and helps TWRP create system reaching conditions again. Question is, if it can be fixed permanently, or without a manual help.

[image]

procount, 2021-12-14:
The following 2 issues have been identified:
1. Lineage19 uses additional ext4 FS features that PINN did not support, 
2. The location of fstab has moved from v18.
These prevented the fstab being updated properly for the TWRP recovery partition.
I am in the process of creating a fix for this which can hopefully be applied retrospectively without having to reinstall Lineage.

Thanks for reporting this.

NumberOneGit, 2021-12-17:
KonstaKang made an update to Lineage19 that fixes a problem that prevented Wifi from working altogether for many users, so if it is possible to use the newer version while making the other updates that would be a massive help.

procount, 2021-12-27:
PINN has been updated to support Lineage19 installation better.
Lineage19 has been updated.

Please let PINN update to v3.7.5 and then reinstall LineageOS 19.

closed 2021-12-27 by procount