procount/pinn #599: Preserve cec_keys.json on update

redm123, 2021-12-13:
Would it be possible to preserve the cec_keys.json during an update? So I don't have to manually put it back afterwards each time. (The first time I was actually lucky to have the file still around locally, because I wasn't aware that it will be overwritten.)

procount, 2021-12-13:
This should be the case now. The latest version does not include cec_keys.json or joy_keys.json, but cec_keys.def and joy_keys.def instead, so it should not overwrite any existing files. These .def files need to be copied to .json files if they are to be edited. If this is not the case please provide more details.

redm123, 2021-12-16:
Ah ok, you are so good! 😁 Well, I admit I just found the item to create this ticket on my TODO list for this system, and didn't check again if it was fixed "just now"... 
On the other hand I also don't see def files on my 3.7.4... But that may be because I had to manually copy it over, as automatic update did not work. Anyway, let's just close it and I'll come back in case I come across this again.

closed 2021-12-16 by redm123