procount/pinn #600: Difficult to install OS

mikuzz, 2021-12-21:
Hello,

first of all thanks so much for the great job, I like so much this approach and all possibilities that I have to use my rpi4.
as second sorry if this is the wrong place to post.

I have several os installed in my SDD included the LineageOS 18.1 Android TV.
By default PINN allow to install the release of March (called 6.3).

Since then, konstakang released other versions that I would like to test (64-bit kernel, some drivers update, a native VNC server, etc)
I tried to install the last version (called 23.11) with manual actions:
- I created 4 new partitions in my sdd with the same size of the original image (sda36 / sda37 / sda38 / sda39 in my case)
- I copied all files and folders from the original image to each partition
- I modified /SETTINGS/installed_os.json to include the new os then I changed the reference for the partition to sda36, etc and the path of the "folder" to a new one
- I duplicated the folder /SETTINGS/os/lineage18-rpi4-atv/ with /SETTINGS/os/lineage18-rpi4-atv2 to be consistent with the previous modification.
- I add and modified correctly the os_config.json in the boot partition

When I reboot, the new os appears in the list but the boot fail: rpi4 reboot after few seconds.
this process is correct? how can i fix it?
thanks in advance

procount, 2021-12-22:
TBH Android is one of the hardest OSes to convert.
What you have done so far isn't wrong, but it is not complete as you need to tell lineageOs where it can find its new partitions, which means modifying fstab files within cpio archives etc.

Fortunately, I am fixing an issue with the lineageOs19 conversion which needs a new version of PINN. I am taking the opportunity to update all the lineageOs variants, so if you can wait a short while it will be done for you.

mikuzz, 2021-12-22:
Thanks so much.
I will wait but I will try anyway to learn a little bit more starting from
fstab, then if you have more details to share I would be happy to study how
android is working.
You can tell me also if using pinnify could be considered as valid
alternative solution?

procount, 2021-12-22:
Reading your actions again, try going to the maintenance menu, select your new os and choose fix up- rerun partition setup. It may fix it for you, but this is where I am fixing an issue.

Pinnify was Sakaki's answer to automating the conversion process for Gentoo. It can be used for similar OSes without much modification, but each OS is different and Android is quite different.

I have in fact used it as the basis for my conversion scripts, but they are not ready for release yet.... one day maybe!

mikuzz, 2021-12-22:
hello procount,
thanks for the suggestion but after the "fix" in the maintenance menu the
situation is the same: I can detect that the boot starts because I can see
the change of the resolution on my tv but after few seconds rpi4 reboot.
I'm looking for any logs to understand where is the problem but I don't
think that android create any log at this stage.

procount, 2021-12-24:
Please try downloading this [os_list_v3.zip](https://github.com/procount/pinn/files/7771944/os_list_v3.zip), unzip it to os_list_v3.json and copy it to the PINN partition of your drive.
When you reboot, you should have a new lineageos18-rpi4-atv OS available that you can use to replace your old version.
Please let me know how this works for you.

mikuzz, 2021-12-26:
Hi procount,
sorry for the late reply.
I confirm you that is working fine.
I added the file in the pinn partition (called settings), reinstalled the
os. After I overwrite the partition userdata with the copy that I had and
everything was fine. Now I reinstalling everything from scratch.
Honestly the improvement is huge, everything is smoother.
Next time I will try again to do it alone, I would like to help, at least
to create images for Android :)

procount, 2021-12-26:
I have updated Lineage18 and 19, so you can delete the os_list_v3.json file from above and install as normal.

closed 2021-12-27 by procount