procount/pinn #603: Prevent PINN from announcing updates

vanfanel, 2021-12-27:
Hi,

I use PINN on stable systems, so having it automatically going into the recovery menu to announce updates once in a while is really annoying.
In other words, I want PINN to always boot to the OS selection menu.

So, is there a way to disable automatic updates detection?

Also, if I apply an update, I have to edit config.txt re-add the disable overlay and disable splash options. With each update, I have to manually do the same.
So, is there a way to store permanent settings that survive PINN updates?

Thanks!

procount, 2021-12-27:
> So, is there a way to disable automatic updates detection?

Yes, use the 'no_update' option. See https://github.com/procount/pinn/blob/master/README_PINN.md#cmdline-options

> So, is there a way to store permanent settings that survive PINN updates?

I did try to use `include "user.txt"`, but it is only suitable for _some_ of the config.txt settings. The only other option is not to overwrite config.txt, but there is the danger that if this file ever needs to change, it would have to be updated manually. However, this has only happened once, so maybe a small price to pay. Will look at it soon.

>  It seems that having a joystick ... is triggering the recovery menu without pressing any keys.

Some joysticks do not have a deadzone, so they continuously generate keypresseses. You may may need to create your own joystick mapping file with appropriate deadzone settings. Please see https://github.com/procount/pinn/blob/master/joy_keys.md

vanfanel, 2021-12-27:
@procount Thanks! I found the cmdline options VERY useful, should have searched for it in the first place...

closed 2021-12-27 by vanfanel