procount/pinn #604: OS not Booting after etching .img of SD backup

jmodra, 2022-01-08:
Hi, 
I backed up my SD card using win32DiskImager so I could use the SD for another pi project. 
I then went to restore the backed up .img using BalenaEtcher but now it wont boot the OS. 

It just sits at the pi bootloader screen
[image]

I stupidly did not create a backup OS of my custom OS inside PINN.

It appears the issue maybe to do with the way the new formatted SD card is partitioned.
As the "`installed_os.json`" lists "`/dev/mmcblk0p6", "/dev/mmcblk0p7`"
While  "`fdisk -l" returns `" 
```
/dev/mmcblk0p1 as W95 FAT 16 (LBA)
/dev/mmcblk0p2 as Extended
/dev/mmcblk0p5 as Linux
```

But I have no idea how to correct. or if that's even the issue.

I have tried running partition_setup.sh and also rebootp 5 (to see if forcing a boot to partition 5 would work). But no luck. 

Any advice much appreciated. Until I go back to a really old PINNS backup I have. 
But done some much customization since.

procount, 2022-01-08:
Are you trying to restore your image to the SAME SD card, or to a SIMILAR SD card?
What size is your img file?
Were there are nay errors when you restored it?
Fdisk is only returning the 3 PINN partitions. It is not showing the partitions belonging to your customised OS. Either they weren't backed up in the image, or they were backed up and restored, but then somehow deleted. This can happen if recovery.cmdline contains `runinstaller`, but I wouldn't expect that if you have just backed up a working PINN SD card. BUt just in case, after restoring the image, just check recovery.cmdline does not include runinstaller before first booting.

jmodra, 2022-01-09:
Yeah it is the same SD (8GB) I backed up and restored to. 
The img file is 7.36GB

The recovery.cmdline file on the SD does not appear to contain `runinstaller`. 

But I think you are right that the OS partition does not seem to be installing. 
As I notice the SD in windows only shows "RECOVERY"

The actual .img file (if I open in 7-zip) shows three files
0.fat - which appears to contain all the PINN recovery files
1.img - which appears to contain the OS (including [SYS] journal & OS (retropie4)
2.img - Some unix file

Thanks for guiding me through this

lurch, 2022-01-12:
Very strange. Win32DiskImager and Etcher (as well as RPiImager) all work by reading and writing the card at the very lowest-level of "raw blocks" which "sit underneath" even filesystems or partitions. So _in theory_ making a `.img` backup of an SD card, and then writing it back to the same card, should produce a bit-for-bit identical copy.

If you're able to boot Linux, you can "loopback mount" your image file (search google for kpartx), which will give you a much better view of what's actually inside your backup image, than a Windows-tool like 7-zip.