procount/pinn #606: Ubuntu 21.10

autodeck, 2022-01-10:
I see that Ubuntu 20.10 desktop is available but not 21.10. How would I go about adding this? Are there any instructions?

procount, 2022-01-17:
There are general instructions in README_PINN.md and a general tutorial in the Wiki, but each OS is slightly different.
I will look at it when I can, otherwise, you can look at what was created for the 20.10 version.

cowboy-rosef, 2022-01-31:
I have tried the 20.10 Mate version and I was able to update it to current, though it took all afternoon with an ssd. I couldn't get reg Ubuntu updated. Since 20.10 was end-of-life last summer and I believe 21.04 just hit EOL, I wanted to ask, if you don't have the time or means to update them, if you would please consider taking them down until you do have time.  It is a security concern at this point, and it might be unwise to leave them up there for too long. Plus, the LTS is just around the corner, you can leave that one up for years. And hey, I'm a  big fan of the project, procount. Thanks.

procount, 2022-01-31:
Ubuntu 21.10 has been added. Please report any issues. If it is ok, I can look to removing the older versions.

cowboy-rosef, 2022-01-31:
That's awesome. I'll put it throught the paces this week. If there's anything drastic, I'll let you know right away, otherwise I'll get back to you with any other feedback later in the week. Thanks man.

cowboy-rosef, 2022-02-01:
I had kernel panics prevent me from booting into ubuntu-mate. Reg ubuntu installed without issue but locked up twice within 10 or 15 minutes. Tried the first time with Wayland and the second time on xorg with the same result. Curious what results others may have had. It's always possible that I set something up wrong.

procount, 2022-02-01:
Which version of UbuntuMate has Kernel panics? I am just converting a 21.10 version that seems to work, although a system problem is often detected, but it doesn't panic. I haven't compared it with a normally installed version.
How do you switch between Wayland and Xorg so I can try it similarly?

cowboy-rosef, 2022-02-01:
I tried both of the versions of Mate. Since I got the same result on both I didn't dig any deeper. Switching to xorg is easy. When you're at the login screen, when you select your user, there is a little gear icon that appears on the lower right part of the screen for switching desktop environments. since they made Wayland the default, the one that just says ubuntu is for wayland. the other one should say ubuntu on xorg.. any other desktop environments that are installed will also be listed. I usually stay in xorg, fewer issues that way.

cowboy-rosef, 2022-02-01:
I'll try running the mate installer again so i can give you exact wording, but it may be a little later.

procount, 2022-03-22:
The method of installing PINN should not really make a difference, unless they installed different versions of PINN. Odd.

cowboy-rosef, 2022-04-03:
that might what happened. not sure when they started to include it, but there is an option these days to download Pinn through the rpi-imager. there's a good chance they may have changed a thing or two, just in the course of incorporating it into their system, i would think.

cowboy-rosef, 2022-04-03:
I just ran rpi-imager from the terminal to see if it indicated something like that, and it is pulling the march 18th pinnlite.img from github. so yeah, I dont' know man. I haven't had any issues since. The only thing I can think of, other than operator error ;) , is could there be a looming issue related to secure boot? It seems doubtful but I have had so much frustration dealing with secure boot on PCs. It requires a usb that has been formatted in a way that no one ever formats a usb, with a large ntfs partition and a small fat32 partion at the tail end of the drive, GPT table. Several odd requirements to actually achieve "Secure Boot". That was the final straw for me with Windows. And lately Ubuntu has been using Secure Boot as a marketing angle on Ubuntu Core, and the rpi-imager is the install method canonical has been pushing. no clue if it's actually related though. Mostly mentioning it because it's got potential to cause real problems.

cowboy-rosef, 2022-04-03:
to give a little context, Microsoft's approach with Secure Boot was to insist that you only use the usb imaging software that they provided, and they offered false reasons as to why, claiming other software tried to write the image too quickly and created errors, when really it was odd partitioning layout that caused the most problems. If you made a USB using say, dd or gnome-disks, you would get error messages that had nothing to do with the actual problem, which was that it was failing the Secure Boot check. I learned all of this the hard way. You can switch Secure Boot into setup mode or off, depending on the motherboard, and bypass this hassle, but I just had to know why it wasn't working. It causes issues in unexpected places, just something to be aware of I guess.

cowboy-rosef, 2022-04-03:
in my ranting i missed the more obvious and likely explanation. I was using a more recent version of Pinn when I installed the rpi-imager, because it was several weeks later. case closed i think.