procount/pinn #607: Blank Screen Problem

Madboyz25, 2022-01-11:
Hi 
I having problems with booting some Operating systems I used the Pinn online tool and a 2 Terabyte HDD The os Im having trouble booting are rpios 64 bit, ubuntu 20.10, and twister os The only Thing I tried that worked so far is lakka the led on keyboard and mouse turn on and see the raspberry pi firmware booting but after that the screen gose blank. I do not know what wrong.

Madboyz25, 2022-01-11:
I also used the pinn-lite that is on sourceforge

procount, 2022-01-12:
I suppose you have no trouble viewing your monitor when you use PINN as you have managed to install several OSes?
I suspect it might be due to the new KMS display drivers in those OSes. Maybe try going back to the FKMS drivers (where F=Fake, not Full - that nomenclature always confuses me!) by modifying config.txt for each OS (you can do that in PINN's maintenance menu).
Does your monitor have a 1366x788 resolution? That can also be a problem with the KMS drivers.

Madboyz25, 2022-01-12:
Ok I will try that thanks my monitor as far as I've seen has not had 1366x768 all Ive used is 1920x1080 but I will try kms in config.txt thanks for helping

Madboyz25, 2022-01-12:
If it works I will close the issue

Madboyz25, 2022-01-12:
I tested it out they wer all on fkms to start I changed them to kms to se if they fix but to no avail I don't know what to try next but do you have anything I can try

Madboyz25, 2022-01-13:
Im just gonna close this issue now

closed 2022-01-13 by Madboyz25