procount/pinn #608: Use PINN from USB Disk to install to SD/MMC

bryceschober, 2022-01-12:
I see from the manual's "[Local Images on SD card" section](https://github.com/procount/pinn/blob/master/README_PINN.md#installing-to-a-usb-device) that

> NOTE: PINN can only install the boot partition of an OS to the same drive that PINN is stored on. So for option 3, the USB drive must have PINN installed on it and the Pi must be booted from this drive.

Is this a fundamental limitation, or a feature in waiting? I am wondering if PINN can be used to image the RPi's SD card or CM4 eMMC when installed on and installing from a USB flash drive.

procount, 2022-01-12:
The RPi can boot from several devices e.g.: SD card, USB drive or eMMC (in the case of CM4). You can control this to some extent through the EEPROM boot order settings in the latest models. If you want to use PINN to choose which OS to boot at startup, then PINN must be the first thing that is loaded from the boot device. Having booted into PINN, the limitation is that PINN can then only specify a boot partition on the same device to load a selected OS from. It cannot select a partition on another device for example. 

Because of the above limitation, PINN will only install OS boot partitions to the same device that it is executing from. So if you want to multiboot several OSes from the eMMC on a CM4, then you must put PINN on the eMMC and execute it from the eMMC in order to install the OSes onto it, even if the OS install files are on a USB stick. 

If you have installed PINN on a USB stick and boot it in a CM4, then it can only be used to install OS boot partitions to the USB drive. (Only the root partitions could be installed to the eMMC in this case) and you would need to boot from the USB drive each time.

Does that answer your question?

bryceschober, 2022-01-14:
Yeah, I was hoping that PINN installed on the USB disk could run an automated O/S installation from the USB disk onto the eMMC, replicating the functionality of CloneZilla for initializing disk images on CM4-based hardware. It sounds like that is not possible, but I'm wondering whether that is a fundamental limitation, or a feature waiting to be implemented by the right person?

lurch, 2022-01-14:
@bryceschober If you're flashing both the eMMC and the bootloader, isn't it easier / more reliable to use the existing rpiboot functionality for this? https://www.raspberrypi.com/documentation/computers/compute-module.html#flashing-the-compute-module-emmc

bryceschober, 2022-01-14:
@procount @lurch Yes it would be easier, and perhaps more reliable... if your target system actually supports OTG. Also, I was hoping that perhaps the imaging speed could be faster also, but now that I write that in front of my eyes, it sounds silly. All that said, I'm still interested in it for other use-cases, so my question remains:

Is this a fundamental limitation, or a feature waiting to be implemented by the right person?

lurch, 2022-01-14:
> Also, I was hoping that perhaps the imaging speed could be faster also, but now that I write that in front of my eyes, it sounds silly.

Not that silly - many years ago I was working on Etcher and we found that Linux (running on the CM3 IIRC) was able to write to the eMMC faster than rpiboot was - which is why https://github.com/balena-os/amber-etcher-kernel was developed (which gets used by https://github.com/balena-io-modules/node-raspberrypi-usbboot).
Although PINN "installs" OSes by creating partitions and extracting tarfiles on top of a filesystem, which will be (slightly?) slower than just a low-level block-by-block write (as might be done by Rpi Imager).

> Is this a fundamental limitation, or a feature waiting to be implemented by the right person?

Yeah I guess there's no fundamental reason that PINN couldn't boot from USB and install itself to eMMC - other than that this goes against a lot of PINN's assumptions, so might require quite a bit of refactoring to implement.

procount, 2022-01-14:
> Yeah, I was hoping that PINN installed on the USB disk could run an automated O/S installation from the USB disk onto the eMMC, replicating the functionality of CloneZilla for initializing disk images on CM4-based hardware.

Could you clarify exactly how you are thinking this would work?
@lurch (and I earlier) have assumed you want PINN on the eMMC so that you can install and select between multiple OSes on your eMMC.
But I wonder with your references to clonezilla and disk images if maybe you are expecting to use PINN, booted from a USB stick, to install a normal single disk image (e.g. "raspberry Pi OS.img") onto the eMMC such that after installation, there is no PINN, no multi-boot, just that one single OS.
Which is closest? Or do you have some other method in mind.

bryceschober, 2022-01-14:
@procount Yep, I want the second one. PINN installed on USB that contains an ordinary image (gzipped, preferrably), auto-installed to the eMMC on boot, and PINN never existing on the eMMC.

procount, 2022-01-14:
I suppose it _could_ be a new feature of PINN, but it's quite a way outside its remit as a MULTI-OS installer, boot manager and OS recovery and administration program. I'm not sure it is something I would be doing because:
1) It's real benefit would be for CM4 users, and I don't think there will be a big market for it.
2) PINN doesn't install image files.
3) I don't have a CM4 to develop this with.
4) I don't know much about CM4 in general and its eMMC requirements
5) Probably better to use RPT developed rpiboot.

lurch, 2022-01-15:
I agree with @procount , PINN only deals with tarball-format OS images, so it makes no sense to try to use it to install raw-disk-format OS images.

You might want to search for a Raspberry Pi version of the CloneZilla tool you mentioned earlier (if such a thing exists), or perhaps you could just install Raspberry Pi OS onto a large USB flash drive, boot it on the CM4, install RPi Imager to the USB drive, download your OS-image-of-choice to the USB drive, and then just use RPi Imager to write the OS image to the CM4's eMMC. Rinse and repeat on other CM4s as desired.
If you want something more automated, perhaps you'll need to script something using Raspberry Pi OS Lite and `dd`, or possibly create a custom solution using buildroot if you want something simpler that boots faster than Raspberry Pi OS Lite? :man_shrugging:

bryceschober, 2022-01-15:
> * It's real benefit would be for CM4 users, and I don't think there will be a big market for it.
> * PINN doesn't install image files.

Um... I'm talking about using partition image files and JSON in an `/os` directory, as documented at https://github.com/procount/pinn/blob/master/README_PINN.md#local-images-on-usb-stick

> * I don't have a CM4 to develop this with.
> * I don't know much about CM4 in general and its eMMC requirements
> * Probably better to use RPT developed rpiboot.

bryceschober, 2022-01-15:
But yes, it does sounds like I'll have to build my own thing if I care enough...

lurch, 2022-01-15:
@bryceschober The OS image files used by PINN (and NOOBS) are very different to the "ordinary image (gzipped, preferrably)" files that you're wanting to use, that e.g. Etcher or RPi Imager would write. It's just unfortunate that the name "image" tends to be used for both.
https://github.com/raspberrypi/noobs/wiki/Standalone-partitioning-explained and https://github.com/raspberrypi/noobs/wiki/NOOBS-partitioning-explained go into more of the details. More succinctly, the PINN-format OS files are in e.g. http://downloads.raspberrypi.org/raspios_armhf/ (you need **all** the files except `piserver.tar.xz`) whereas the RPi-Imager-format OS files are in http://downloads.raspberrypi.org/raspios_armhf/images/raspios_armhf-2021-11-08/ (only the `2021-10-30-raspios-bullseye-armhf.zip` file is needed).

bryceschober, 2022-01-15:
OK, thanks for all your pointers and patience! :wink:

closed 2022-01-15 by bryceschober