procount/pinn #610: Issue recognizing sda in "Download space"

NumberOneGit, 2022-01-17:
Under "Install space" I have no issue recognizing my USB drive and installing/re-installing all OS's.  However, I tried to re-install RPOS Buster 64 bit after a failed attempt to upgrade to Bullseye and the pull-down is not showing my drives at all.  Is there a known issue with "Destination drive" options not showing up in "Download space"?  I will wipe my drive and attempt to re-install PINN and see if I have the same issues.

On a side note, it appears that many users are having issues when trying to manually update to Bullseye directly from Buster and a fresh install is recommended.  Is there a plan to include these repos in PINN in the near future for a fresh install?  If not, I will try to figure out how to manually install them.  Thanks!

procount, 2022-01-17:
Sorry, I'm not quite clear what you are trying to do. 
Are you trying to download RPOS Buster64 to your USB drive (so that you can later re-install it), but your USB drive is not showing up in the Destination Drop down?
Remember that you must have a /os/ folder on your USB drive for it to be recognised.
You could always reinstall from the internet instead of from USB 🤷‍♂️

I guess my 64 bit image is a bit out of date. I shall have to update it to the latest version.

procount, 2022-02-09:
Raspberry Pi OS 64 is now available.