procount/pinn #611: Blank screen

Bhavyansh03, 2022-01-19:
I couldn't find any solution for blank screen
I tried using fkms drivers but those were also unsuccessful

procount, 2022-01-19:
What blank screen? 
Is it with PINN or with the OSes it has installed?
Is this the same issue as #607?

What monitor do you use? 
What connection? 
What resolution?
What model of PI? 
What version of PINN?

Bhavyansh03, 2022-01-19:
It's with pinn not with os I haven't installed any os yet
Monitor- Aopen 22CV1Q
Connectio's with pinn not with os I haven't installed any os yet
Monitor- Aopen 22CV1Q
Connection - hdmi(closer to type c port)
Resolution - 1920x1080
Model - pi 4B 2gb
Pinn - latest versionn - hdmi(closer to type c port)
Resolution - 1920x1080
Model - pi 4B 2gb
Pinn - 3.7.5
{I am using pinn after a long time it used to work before I don't know which version it was without any problems}

procount, 2022-01-19:
Is this a new Pi4 2GB and the first time you are using it?
Have you ever got a picture on your screen yet?

Without any SD card installed, just plug it into your monitor and power supply. You should get the red power LED lit on the RPi and the display should show a generally black screen at 640x480 @60Hz with a raspberry logo in the top left of the screen, a QR code in the top right, some other information and a request to insert an SD card.
Try the other HDMI port (the above screen should appear on both ports).

After you get to this stage, then you can add the PINN sd card (with the power off). It should start with the familiar rainbow square pattern, then a small green square in the middle with the mouse cursor and finally the PINN screen.

Bhavyansh03, 2022-01-19:
Right now also it's running chromium os

procount, 2022-01-19:
You need to establish if it is a screen issue, or a non-booting issue. 
Does the green LED continue to flash in any way once you get to the blank screen?
Have you tried re-installing PINN to your SD card?

Can you connect it to a network? If you can plug Ethernet into it, find its IP from your router or a tool like Fing, you can add "forcetrigger ssh vncshare" to the end of recovery.cmdline (make sure it is only 1 line of text - don't add extra lines) and see if you can ssh or vnc into it. Or even just a simple PING would establish that it has booted and it is just a screen issue.

Other than that I'm not sure how I can diagnose the issue. If chromium works, and other OSes work, then PINN should work too. Maybe it was just a bad flash to your SD card?

Bhavyansh03, 2022-01-19:
Actually its booting according to me because I can go in to recovery menu using Ctrl + alt +F2
I will try a older version of pinn

procount, 2022-01-19:
Nor me. The firmware & kernel has been bumped a few times since 3.5.4.
I can only suggest you try and narrow it down to the first version of PINN that fails.
Could you try versions 3.5.5, 3.5.6, 3.6, & 3.7?
You can find past versions in https://sourceforge.net/projects/pinn/files/archive/

procount, 2022-01-20:
So the recovery shell works, but ctrl-alt-f1 doesn't show a screen? Very odd.
You could try modifiying config.txt to add:

hdmi_group=2
hdmi_mode=28

Just to see if a different resolution will fix it. 🤷‍♂️

yav12, 2022-01-23:
I too receive this problem with the latest version of PINN on a raspberry pi 4b 4gb.
A black screen shows up and nothing else (after the rainbow splash)
As suggested, I tried older and older versions until one worked, which was version 3.5.5

procount, 2022-01-23:
The only significant change between 3.5.5 and 3.5.6 is the rpi-firmware and rpi-wifi-firmware.
@lurch - Any ideas what changed that could cause this?
I mostly test on Pi4 4GB or Pi400 without issue.

lurch, 2022-01-23:
I'm afraid I know nothing about Pi firmware now - most of my efforts are around documentation, build-systems, and Pico-related things! ping @XECDesign

Tman564wsDEV, 2023-08-12:
I think I fixed it,
1: Download both pinn-lite-3.5.4 and the newest (At this time its 3.8.7)
2: Create a file (To keep track) named pinn-mixup
3: Copy all the files from the newest and delete recover4 and fixup4rc
4: Copy recover4 and fixup4rc to the pinn-mixup
5: Copy all the pinn-mixup files on to the DOS FAT or FAT32 SD Card
6: Enjoy :)
ps: This worked for me so it should work for you

Tman564wsDEV, 2023-08-12:
Ok nvm now no os work, I will work on it.

Tman564wsDEV, 2023-08-14:
Got it agen, use the imager

PunishMints, 2023-11-28:
I also had the same exact issue with a Raspberry Pi 4 Model B running pinn-lite latest, pinn-lite 3.5.5 works without any issues.