procount/pinn #612: Add option to change UI of boot selection menu

superpupervlad, 2022-01-19:
I don't really care about UI in PINN and don't want to redesign it like in [this](https://github.com/raspberrypi/noobs/issues/610) issue because once you set it up you will rarely return to it. But menu with selection OS to boot is seen every boot and it's not very good looking. Also I believe there are a lot of people like me that use RPi as TV box with LibreELEC/RetroPie/Android TV installed and this menu poorly optimized for TV screens. It would be ideal to do something like this:
[image]
(burg bootloader theme)

As far as I can tell it is possible to achieve with [QListView::IconMode](https://doc.qt.io/qt-5/qlistview.html#ViewMode-enum) in [bootselectiondialog.ui](https://github.com/procount/pinn/blob/master/recovery/bootselectiondialog.ui) and changing all down and up buttons commands to right and left. 

Also it would be great to customize width and height of window (or even make it fullscreen) in some config file but I don't know anything about Qt so I'm not sure is it even possible to do this after compilation.

procount, 2022-01-20:
I don't think it is quite as simple as just flipping the viewMode to IconMode. It would probably need to be part for a complete UI rewrite.

If you have a mouse, try double-clicking on the title bar, or click and drag one of the window corners.

DerangedPineapple, 2022-01-23:
That would be magnificent if OS selection screen could be themed like grub can be (I do understand there are no similarities technically between what pinn offers and grub). Currently that screen is a bit hard to use via controller/joystick or tv remote via hdmi cec.

Also from 3 meters away with 64 inch udh screen everything is rather small (I assume font sizes can be changed currently).

Other thing is that my family would be using libreelec and retropie and switching between those is a bit cumbersome as switching OS by using tv remote is slow when you are actually controlling the cursor via remote's "arrow" keys instead of controlling the selections in the OS screen.

I know this is a tall order and it might not even be feasible to add theme support for the boot screen. Anyways, just wanted to give a thumbs up to this idea.

lurch, 2022-01-23:
> It would probably need to be part for a complete UI rewrite.

Rather than rewriting PINN's UI entirely, I wonder if another option might be to add a `os_selector` option to PINN's `cmdline.txt` file; **if** that option is set (it would be unused by default) then when PINN gets to the OS-boot-selection dialog it would just quit the existing `recovery` program, and run the command specified by the `os_selector` option instead.
That way none of the PINN UI has to change, and people can write their own OS-selector programs entirely independently of PINN (just making use of the same PINN JSON files) that look and function however they desire. Of course whether any external developers would be willing to put the effort in to actually write an OS-selector program is another question entirely... :thinking: :man_shrugging:

lurch, 2022-01-23:
(P.S. one drawback to the screenshot shown by @superpupervlad is that it *only* shows the OS-icons, not the OS-names. So for new users who might not be familiar with the different icons used by different OSes, it might be hard to figure out which OS is which? Just another example of "you can't please all the people all the time" :wink: )

jjlankinen, 2022-01-24:
I hardly think that the provided screen shot ever ends up being the final theme, I consider it as an idea to be build upon. 

Adding OS names/titles beneath the thumbnail should be trivial IF that kind of theme would be even possible to be added to the boot selection.

It would be nice also if exposing the theming of OS selection to the users would be possible.

procount, 2022-01-24:
These are all good suggestions. 
I know the UI is not ideal, but since I also know many people do not read documentation, I'll point out a few things you can do within the current restrictions of the UI to customise it, that you may not be aware of:

- The background screen colour can be changed
- The background wallpaper image can be changed and stretched to fill the screen.
- The windows style can be changed between Windows, Motif and Platinum
- The windows can be resized by dragging, or maximised by double-clicking on their title bars.
- The font size can be changed by using the +/- keys (these shortcut keys are only operational on the installation menus, but their effect is still visible on the boot selection dialog)

IIRC, the last 2 are not persistent across boots, but that is something I could add, and they're probably not so easy to change with your TV remote. They were an attempt to go some way towards making the large TV experience more functional, yet maintaining the original NOOBS architecture.