procount/pinn #613: Install GApps on Lineage 19 ?

MopheusDG, 2022-01-20:
Hi, I'd like to know if it's possible to install NikGApps or BiTGApps someway using Lineage OS 19 under PINN. I'm using an SSD with multiple OSes but when I boot Lineage on TWRP and try to install the GApps I get an error when mounts the DATA drive/folder and the instalation fails and can't boot anymore.

Is it possible to install it or am I doing something wrong ?
Thanks.

procount, 2022-01-21:
I don't know anything about NikGApps or BiTGApps, but I did make sure that GApps could be installed according to Konsta's FAQ for LineageOS 18. 

LineageOS 19 looked like it operated in exactly the same way, but a change to its file system meant that the setup script didn't work correctly. This required an upgrade to PINN to support the new file system attributes and a slightly modified setup script. (See https://forums.raspberrypi.com/viewtopic.php?p=1954356#p1954356)

You need to use PINN v3.7.5 and install the latest LineageOS 19. That should then work.

MopheusDG, 2022-01-21:
I have updated PINN to the latest version via the auto-update function. Now my doubt is, can I just re-install only the Lineage OS and that should be enough ? By the way, there's a new updated version on the site: 14.1. 2022, just FYI. Cool new updates as always :) and thanks for the reply.

procount, 2022-01-21:
Reinstalling Lineage _should_ work. If it doesn't please let me know.
Re-installing will also get you the 14.01.22 version as that is the version I update it to :)

MopheusDG, 2022-01-21:
Great, will try later tonight and post back with some screenshots if I run into problems to help. Thanks @procount.

procount, 2022-01-21:
Hopefully it will work this time. IIRC, just re-installing this latest version of LineageOS19 with its new script may have worked on its own. The update to PINN was necessary to make the "re-run partition script" work because once LineageOS has been booted for the first time, it changes some attributes of its filesystem so that PINN can no longer mount one of its partitions.

MopheusDG, 2022-01-22:
Hi @procount not sure if I'm doing something wrong or what, but I tried re-installing and replacing the Lineage and both times I cannot boot into Andriod, I don't even see the main screen.

Usually I get no signal from HDMI if I use vc4-kms-v3d driver, but after switching to vc4-fkms-v3d driver used to work fine...

Anyway I can check for errors or something ?

Thanks.

MopheusDG, 2022-01-22:
Finally I was able to install Google Play nicely, the problem was when I was trying to use a USB Pendrive for the .zip file, it seems using another USB Pendrive with an external USB SSD was generating some kind of conflict probably. I recommend everyone to download the file directly from Lineage and then installing from the local drive, not using another drive just in case...

Thanks @procount for your support.

procount, 2022-01-22:
Glad it works for you now.

closed 2022-01-22 by procount