procount/pinn #616: Git merge conflict has been inadvertently committed

lurch, 2022-01-23:
See https://github.com/procount/pinn/blob/master/recovery/translation_zh_TW.ts#L842-L859
This results in `lrelease error: Parse error at .../pinn/recovery/translation_zh_TW.ts:842:2: Unexpected '<'.` when you try to run `updateqm.sh`.

procount, 2022-01-23:
How did that go unnoticed?

(How come you are doing some late night building?)

lurch, 2022-01-23:
Also, https://github.com/procount/pinn/blob/master/README_PINN.md#run-build-script says "If your build machine also has some QT5 components, it is useful to export QT_SELECT=4 before building to ensure the QT4 component versions are selected." so perhaps [this line](https://github.com/procount/pinn/blob/master/recovery/updateqm.sh#L3) ought to be removed? Or is that only causing me a problem because I was trying to run `updateqm.sh` on Ubuntu 16.04 ?

lurch, 2022-01-23:
> (How come you are doing some late night building?)

PINN is a project that I know takes **ages** to build, so I thought I'd use it to "benchmark" my (new-ish) laptop against my (old-ish) desktop :laughing:

procount, 2022-01-23:
Not sure I ever run `updateqm.sh`, but that line looks errant. I moved development from Ubuntu 14.04 to 20.04 now to match RPF build server.

lurch, 2022-01-23:
Probably not worth "fixing" if it's only old Ubuntu versions that are affected.

procount, 2022-02-09:
Fixed by 3.8

closed 2022-02-09 by procount

lurch, 2022-02-11:
https://github.com/procount/pinn/blob/master/recovery/translation_zh_TW.ts#L842-L859 still looks wrong?

procount, 2022-02-11:
I should probably push the commit up, I suppose..... :wink: