procount/pinn #617: Kindly update Lineage OS builds.

saumyashhah, 2022-01-24:
Hi there, kindly update lineage os builds of konstakang.  
(Lineage 18 ATV)

procount, 2022-01-24:
I've only just updated them and now he has another 6 releases - I can't keep up! :wink:
OK I'll see what I can do....

lurch, 2022-01-24:
Perhaps it's worth trying to convince the Lineage developer to produce "PINN compatible" builds himself, since there's clearly quite a large demand for it...?

saumyashhah, 2022-01-24:
> I've only just updated them and now he has another 6 releases - I can't keep up! 😉 OK I'll see what I can do....

Yeah, kindly update Twister too,

saumyashhah, 2022-01-24:
> Perhaps it's worth trying to convince the Lineage developer(s) to produce "PINN compatible" builds themselves, since there's clearly quite a large demand for it...?

Yeah, he told to ask PINN dev, stuck in a ∞ loop lol.

procount, 2022-01-31:
Lineage 17/18/19 updated. Please report any issues.

saumyashhah, 2022-02-01:
Hi, did you update the main files? 
I see only some jsons are updated.

Will surely report issues.

procount, 2022-02-01:
Yes.
Where are you looking? 
I'm experimenting with a more seamless upgrade process so that I can easily roll back if there are any problems.

procount, 2022-02-01:
Maybe you are looking the wrong place. They should all be current, at least for now ;)

os/17 is the only November dated one.
os/18 is Dec 21
os/19 is Dec 21
os_next/17 is Dec 21
os_next/18 is Jan 22
os_next/19 is Jan 22

[image]

If the new ones are ok, I will migrate them to the os folder

saumyashhah, 2022-02-01:
Okay,
Will test LOS18 and tell you in some days, so I can directly download them from Pinn on pi4? or have to do something special?

procount, 2022-02-01:
Yes you can download them directly from PINN.
Although, some people are reporting some bad transfers from the sourceforge hosting site at the moment. 
If you experience errors, try installing multiple times. Or even downloading them to a separate USB drive via PINN or manually (see the README_PINN.md file) and install them from a local copy if you get persistent errors.

mikuzz, 2022-02-01:
Hi procount,
yesterday I was able to install Lineage 19 (14.01.2022)
The installation was fine as usual, I didn't get further in the
installation of gapps and all others apps for the moment.

procount, 2022-02-09:
Closing now the latest LineageOS's are available and PINN is updated to back them up properly

closed 2022-02-09 by procount