procount/pinn #618: VNC doesn't work

Chris-Idema-WizzDev, 2022-01-25:
I have a raspberry pi 4 connected with an ethernet cable to my ubuntu laptop. I have installed pinn-lite successfully before with 2 raspberry pi OS images on the same SD-card. Now I want to repeat the process headless as I have to do it without a monitor now. 

I put the pinn files on a formatted SD-card, added `vncinstall` and `forcetrigger` to recovery.cmdline and powered on the Pi with this SD-card. 
No device shows up when I scan for devices using nmap, so I don't know what IP-address to try. Does pinn have a hostname?
When I use raspberry pi imager I can use SSH and VNC perfectly fine directly with the image, so I know the pi, sd-card and network are working.

procount, 2022-01-25:
Have you connected your RPi _directly_ to your Ubuntu laptop with a single ethernet cable?
Or have you connected both the RPi and your laptop to a router?

PINN requires the network to provide it with an IP address using DHCP, so if you have connected both devices to your network router then you should be good to go. Normally you can view the client IP addresses handed out by your router's DHCP server using your router's webpage (or nmap or fing or similar as you have tried).

If you have connected your laptop and RPi together directly, then your Ubuntu Laptop needs to be running a DHCP server to allocate an IP address to PINN. There are plenty of tutorials available on this. Make sure you allocate a static IP address to your Ubuntu network adaptor that will be reachable from the range of addresses your DHCP server will allocate to the PI.

Hostnames are used to lookup a device's IP address, so an IP address still needs to be allocated first.

Personally, I prefer to use `vncshare` rather than `vncinstall` because it will work on both monitor and vnc without having to keep changing it. I also add `ssh` becuase I often connect to PINN that way for maintenance purposes.

How did you try to set a fixed IP-address? What was it? What was the IP address of your Ubuntu laptop?

lurch, 2022-01-25:
> If you have connected your laptop and RPi together directly, then your Ubuntu Laptop needs to be running a DHCP server to allocate an IP address to PINN.

You'd also need to configure IP-forwarding on your laptop (and maybe DNS proxying too), so that PINN is able to access `http://raw.githubusercontent.com/procount/pinn-os/master/os/repo_list.json`.

Chris-Idema-WizzDev, 2022-01-26:
I have the pi connected directly to the laptop with an ethernet cable.
I used this guide, which worked great for sharing internet connection with my pi: https://nrsyed.com/2021/06/02/raspberry-pi-headless-setup-via-ethernet-and-how-to-share-the-host-pcs-internet-connection/
I used these settings:
```
nm-connection-editor
+
ethernet
name: pi
General/Connect automatically = false
ethernet device (maybe not needed)
ipv4 settings/method = shared by others
ipv4 settings/address = 192.168.4.1
ipv4 settings/netmask = 24
ipv4 settings/Gateway = 192.168.4.1
save

nmcli connection modify pi ipv4.never-default true

disable VPN
```
I also tried different methods such as DHCP and manual.

procount, 2022-01-26:
I followed your link.
That article is for Raspberry Pi OS and needs adapting for PINN
It does not use DHCP, but assigns a static IP address (192.1682.4.2) to the Pi.
The location of that file `/etc/dhcpcd.conf` is not available on PINN, so where did you put it? Did you put it on the rootfs partition of Raspberry PI OS? 

On PINN, using that dhcpcd.conf file should still work, but you need to put it on PINN's root partition (where you will find recovery.rfs). PINN will copy it to /settings on boot up.

Obviously, please bear in mind that using this static IP address technique will get it working with your laptop, but if you try to plug it into a router in the future, it will no longer work with DHCP and will always have the IP address of 192.168.4.2

Chris-Idema-WizzDev, 2022-01-26:
I was able to get it to work. The problem was the network timeout was too short and once another network configuration was used before it would not reconfigure when I switched network configuration.

Format SD-card according to https://raspberrytips.com/raspberry-pi-dual-boot/

Add to recovery.cmdline:
`vncshare ssh forcetrigger networktimeout=0`

To configure ethernet on laptop:
```
run nm-connection-editor
click +
connection type: ethernet
name: pi
General/Connect automatically = false
ethernet device (maybe not needed)
ipv4 settings/method = shared by others
ipv4 settings/address = 192.168.4.1
ipv4 settings/netmask = 24
save
disable automatically connecting to ethernet of other configured network connections.
close nm-connection-editor

nmcli connection modify pi ipv4.never-default true
```
Disable your VPN.

Plug in SD-card and ethernet cable. Power on and enable the network configuration.
`nmap -sP 192.168.4.0/24` Mine reported 192.168.4.165
For ssh: `ssh root@192.168.4.165` pw `raspberry`
For VNC: I use Remina with `192.168.4.165`

procount, 2022-01-27:
hostname should be recovery.lan
Can you try downloading projectspace again? probably it was a temporary internet error

closed 2022-01-27 by Chris-Idema-WizzDev