procount/pinn #619: Nothing Installs.

Noahbx, 2022-01-26:
I was trying to install a new installation of pinns, and then when I type delete and it prepares I get 6-8 errors saying failed to download meta file, after reformating my sdcard using a laptop with a fresh install of pinns it still gives me the same message and I can't install any oses

lurch, 2022-01-26:
> failed to download meta file

Problems with your internet connection? Or a proxy or something?
What version of PINN? Is it connected to your network via ethernet or wifi?

procount, 2022-01-26:
Sometimes Sourceforge has problems, but I've just managed to download and install PIOS-lite (sic)  and Void, so no problems at the moment.

BTW @lurch & @XECDesign - I just spotted this -> https://www.bleepingcomputer.com/news/security/lets-encrypt-is-revoking-lots-of-ssl-certificates-in-two-days/ and hope we're not going to get more SSL errors!

lurch, 2022-01-27:
> hope we're not going to get more SSL errors!

That article says it only affects certificates issued within the last 90 days, so presumably anyone who's had a certificate issued that recently will have no problem replacing it with a newer valid certificate? :shrug:

Noahbx, 2022-01-29:
> > failed to download meta file
> 
> Problems with your internet connection? Or a proxy or something? What version of PINN? Is it connected to your network via ethernet or wifi? EDIT: And what model of Pi?

RPI4 latest ver as of today ,no I've tried download the same files from the same links on my phone and there's no problem

godzinilla, 2022-02-01:
I have enforcing this issue too today. It is a little chance to install any of the os, and literally no chance to install more than one os in a single installation (including the ProjectSpace).

Example of an error:

`Error downloading meta file http://deac-riga.dl.sourceforge.net/project/pinn/os/raspios64/marketing.tar`
`Error downloading meta file http://deac-riga.dl.sourceforge.net/project/pinn/os/raspios64/os.json`
`Error downloading meta file http://deac-riga.dl.sourceforge.net/project/pinn/os/ProjectSpace/os.json`

Looks like no access to these files.

procount, 2022-02-01:
This looks like a sourceforge error.
I experienced something similar, which incidentally was also using the deac-riga mirror, but it was just temporary.

procount, 2022-02-01:
I have raised a ticket with SoiurceForge.

godzinilla, 2022-02-01:
> I have raised a ticket with SourceForge.

Thanks. Can images be download from the sourceforge and loaded to the PINN directly (multiple images)? Or we need to wait for sourceforge to fix this issue?

procount, 2022-02-01:
https://github.com/procount/pinn/blob/master/README_PINN.md#download

godzinilla, 2022-02-01:
> https://github.com/procount/pinn/blob/master/README_PINN.md#download
> 
> EDIT: Hmm, that link doesn't always point to the correct place, so instead try: https://github.com/procount/pinn/blob/master/README_PINN.md#archival-menu

Not sure why but I cannot select my USB drive or sd card in sd reader under Archival menu. Both can be selected under Main Menu but on under Archival...

procount, 2022-02-01:
You must create an `os` folder on that drive. Make sure it is FAT32 or EXT4 formatted. I think NTFS will also work but is not preferred.

godzinilla, 2022-02-01:
> You must create an `os` folder on that drive. Make sure it is FAT32 or EXT4 formatted. I think NTFS will also work but is not preferred.

Of course, I am a total moron today, didn't read the whole instruction ;) Downloading in progress, or it looks like it is downloading - no progress bar, all elements in window grayed out.

procount, 2022-02-01:
You can download all the files for each OS manually from sourceforge to your USB. You should be able to see the required folder structure on the USB now. When you select a file to download from their webpage, there is a button called "Problems Downloading?" Here you can change the mirror you download the files from. Maybe you will have more luck if you choose a mirror close to you.

godzinilla, 2022-02-01:
> You can download all the files for each OS manually from sourceforge to your USB. You should be able to see the required folder structure on the USB now. When you select a file to download from [...]

No option to download all the files in the directory?

procount, 2022-02-01:
I don't think so. That's one reason why I added the download feature in PINN. PINN shoudl autoresume though. Maybe you could try it multiple times to try and get a different mirror?

godzinilla, 2022-02-01:
The problem is connected with not accessible files by the sourceforge side. Let's check for an example the LineageOS 19 ATV:

`https://sourceforge.net/projects/pinn/files/os_next/lineage19-rpi4-atv/slides_vga/`

these files are not accessible from any mirror. The same for `lineage19-rpi-atv.png`.

Maybe these files are simply not exists on the server(s)?

procount, 2022-02-01:
I tried it and failed to download 1.jpg.
But then I changed the mirror to Net Cologne and it downloaded. So I think it is intermittent. or mirror based.

procount, 2022-02-01:
You need to look on the NOOBS server for that one. PINN uses 4 different repos.
https://downloads.raspberrypi.org/raspios_lite_armhf/

godzinilla, 2022-02-01:
> You need to look on the NOOBS server for that one. PINN uses 4 different repos. https://downloads.raspberrypi.org/raspios_lite_armhf/

Thanks. So I have finally installed all my images. Installation passes without problems but there is no any system available in the boot menu, even under the Maintenance menu. All partitions were created and files are available (under the windows).

procount, 2022-02-01:
Sourceforge have removed a mirror from their rotation so hopefully that will help with internet installation.

If there are no systems in the boot menu, then something must have gone wrong somewhere.
Can you post a copy of `installed_os.json`.?
It is on /dev/mmcblk0p5 which is mounted as `/settings` in PINN's shell.

godzinilla, 2022-02-01:
> Sourceforge have removed a mirror from their rotation so hopefully that will help with internet installation.

So they will fix it?

> If there are no systems in the boot menu, then something must have gone wrong somewhere.
> Can you post a copy of `installed_os.json`.?
> It is on /dev/mmcblk0p5 which is mounted as `/settings` in PINN's shell.

File is empty.

godzinilla, 2022-02-01:
[image]
[image]
[image]

This is the structure of /os folder on my USB drive from which installation process were begun.

I have literally downloaded all files from the SF to the corresponding folders.

For the ProjectSpace folders all files were copied from one directory to others.

procount, 2022-02-01:
By removing the faulty mirror, hopefully sourceforge have fixed it, unless you are telling me you still have problems.

if installed_os.json is empty, then PINN does not think you have installed any OSes.

It looks like you have downloaded the OS files correctly to your USB drive.
After downloading them, was the icon to the right of each OS in PINN showing the USB symbol instead of the ethernet symbol, indicating the OS is sourced from the USB drive? Did you then install from the USB drive to your SD card?
Having no `/settings/installed_os.json` suggests you haven't, or there was some other error.
You haven't included anything about what was installed to your SD card or its structure, so it's difficult to diagnose. You can't do this from Windows. Have a look at https://github.com/procount/pinn/wiki/Troubleshooting

godzinilla, 2022-02-02:
> By removing the faulty mirror, hopefully sourceforge have fixed it, unless you are telling me you still have problems.
Not testes yet, still I messing up with my downloaded images.

> It looks like you have downloaded the OS files correctly to your USB drive. After downloading them, was the icon to the right of each OS in PINN showing the USB symbol instead of the ethernet [...]
Yes and yes.

I made some testing and found that issue is somehow related to the LinegaeOS19 ATV image. If you made a single install with ATV and ProjectSpace (one space) both images are installed, ATV is bootable. Everything works as expected.

godzinilla, 2022-02-02:
Finally, after reformating my SD card, downloading images using the PINN (it is working now) and installing in single installation - everything working as expected.

So looks like my issue comes from incorrect, manually downloaded images and/or messed up sd card

.[image]

procount, 2022-02-02:
Glad it's sorted for you.
Feel free to close this issue.

closed 2022-02-09 by procount