procount/pinn #620: piCorePlayer for RPi 4B - request for an image

godzinilla, 2022-01-26:
Hello. Would anyone be so kind and create an piCorePlayer 8.1.0 64 bit image for a RPi 4B? Thank you!

godzinilla, 2022-01-27:
The image available on the server (showall) is not working on the RPi4B and RPi3B.

procount, 2022-01-27:
That image is quite old. It probably predates the 4B, and maybe even the 3B.
I hope to have a new version up soon once I've converted and tested it (along with some other OSes I am doing)

godzinilla, 2022-01-27:
> That image is quite old. It probably predates the 4B, and maybe even the 3B. I hope to have a new version up soon once I've converted and tested it (along with some other OSes I am doing)

Would be great! I have trying to manage it using the BerryBoot too - still without success. I hope that these images are capable to run as multiboot....

procount, 2022-01-31:
piCorePLayer updated for 32 and 64 bit 8.1.0 versions. Please report any issues.

godzinilla, 2022-01-31:
@procount man, thank you! It botted correctly ;) The 64 bit version on RPi4B. Question to you: If I have a fully configured pcp instance on another sd card (exactly the same version) can I somehow copy all of my configuration to the PINN version? I believe it is related to below article:

https://github.com/procount/pinn/blob/master/README_PINN.md#how-to-create-a-custom-os-version

For me it looks quite hard to achieve. Is it so?

procount, 2022-01-31:
I don't know anything about piCore, how it stores its files or how it is configured.
Following the link above is one way to do it - it is essentially what I did to create the new image, although it has to be customised slightly for each OS. 
Additionally you should take a look at  https://github.com/procount/pinn/wiki/How-to-Create-a-Multi-Boot-SD-card-out-of-2-existing-OSes-using-PINN if you want to go down this route.
You will need a Linux based OS on which to do this conversion. Raspberry Pi OS on a PI4 could do it, but typically you need lots of available storage space to manipulate the files.

If you're just talking about copying your music collection, then maybe try copying the contents of your 2nd partition of your configured card to wherever the 2nd partition of your PINN version of piCore is (see installed_os.json on the settings partition - /dev/mmcblk0p5). This would be easier if it works.

godzinilla, 2022-01-31:
> I don't know anything about piCore, how it stores its files or how it is configured. Following the link above is one way to do it - it is essentially what I did to create the new image, although it [...]

No, the piCorePlayer should be configured by user - the addons, shares, names, devices, general settings. And the LMS - Logitech Media Server - is a part of the piCorePlayer. It has it's own configuration. The LMS need to be installed from the piCorePlayer webGUI (there is a dedicated section for this).

I will try the mentioned solution, but If I failed - would you be so kind and create the PINN image using my SD Card image? Just for me.

procount, 2022-01-31:
Still, I don't think you have anything to lose by trying to copy the 2nd partition. If it doesn't work, just use PINN to reinstall piCore again. You just need a USB SD card reader so you can access both cards at once. Let me know how it goes.

godzinilla, 2022-01-31:
So the copying ROOT partition files from the PCP sd card (32 GB, partitions: BOOT 64 MB, ROOT 2 GB, rest unallocated) to the PINN sd card (4 GB, the BOOT partition has the same size, but the ROOT is greater than 2 GB - whole rest of available space on the sd card).

Not sure if I done this correctly: there was a `tct` folder on the source ROOT partition, and the `tct` on the destination ROOT partition. I have removed the `tct` from the destination partition (`sudo rm -r`) and copy the `tct` from the source partition to the destination partition (`cp -r`). System starts with information that `Checking for boot files on mmcblk0p1... Trying to mount /dev/mmcblk0p1. newcp.cfg not found on mmcblk0p1`. There are no any warnings/error in the shell. The webGUI of piCorePlayer and the LogitechMediaServer works correctly, all my settings were persisted. It looks like it is working OK :)

What do you think about it?

I have an additional question: if I want to prepare my SD card for future installation of other systems without loosing my currently configured piCorePlayer system should I use the Project Spaces solution:

https://github.com/procount/pinn/blob/master/README_PINN.md#project-spaces

Or for each new installation the image ot an existing system should be created in PINN then recreated while installing other systems using this image?

procount, 2022-01-31:
> Not sure if I done this correctly:

Me neither. You just have to experiment with it. 
Do the messages about checking for boot files on mmcblk0p1 appear on the original PCP card, or on the originally PINN installed version? Maybe they are benign. I don't know wnough about how PCP works I'm afraid.

Yes, It's a good idea to install one or more ProjectSpaces to reserve space for future OSes. You can replace them with another OS later. Bear in mind that they are only suitable for use with standard 2 partition layout OSes, like Raspberry Pi OS. You won't fit a 4 partition Android OS into a ProjectSpace, for example.

If you later want to add more OSes, then you could use PINN to take a backup of each OS to a USB drive, then use PINN to install the backups to a new SD card along with any additional OSes. Always test your backups work first.

godzinilla, 2022-01-31:
> > Not sure if I done this correctly:
> 
> Me neither. You just have to experiment with it. 
> Do the messages about checking for boot files on mmcblk0p1 appear on the original PCP card, or on the originally PINN installed version? Maybe they are benign. I don't know wnough about how PCP works I'm afraid.

I will check and go back to you soon.

> Yes, It's a good idea to install one or more ProjectSpaces to reserve space for future OSes. You can replace them with another OS later. Bear in mind that they are only suitable for use with [...]

Can it be used without PS or along with the PS, as a regular system selectable to boot? I was thinking about getting the some Android TV image (if available) to other partition.

> If you later want to add more OSes, then you could use PINN to take a backup of each OS to a USB drive, then use PINN to install the backups to a new SD card along with any additional OSes. Always test your backups work first.

Correct me if I am wrong: if I create eg. 4 PS partitions I won't be able to place the single, additional OS there without reinstalling other existing oses? This is how it works on BerryBoot.

procount, 2022-01-31:
> Can it be used without PS or along with the PS, as a regular system selectable to boot? I was thinking about getting the some Android TV image (if available) to other partition.

Yes.
LineageOS-ATV is available in PINN. Being Android it uses 4 partitions so cannot be installed into a standard ProjectSpace.

> Correct me if I am wrong: if I create eg. 4 PS partitions I won't be able to place the single, additional OS there without reinstalling other existing oses? This is how it works on BerryBoot.

Each ProjectSpace reserves 2 partitions: a small FAT32 boot partition and a variable-sized ext4 root partition. 4 ProjectSpaces would therefore reserve 8 partitions. 

Reinstalling or Replacing an OS with PINN does not alter the layout, number or sizes of the existing partitions - the new OS must match the existing partition structure. Adjusting partitions requires more sophistication. 

Berryboot works entirely differently and uses readonly squashfs file images and therefore doesn't mess about with partitions.

godzinilla, 2022-01-31:
> Yes.
> LineageOS-ATV is available in PINN. Being Android it uses 4 partitions so cannot be installed into a standard ProjectSpace.

Thanks, I will check this.

> > Correct me if I am wrong: if I create eg. 4 PS partitions I won't be able to place the single, additional OS there without reinstalling other existing oses? This is how it works on BerryBoot.
> 
> Reinstalling or Replacing an OS with PINN does not alter the layout, number or sizes of the existing partitions - the new OS must match the existing partition structure. Adjusting partitions requires more sophistication. 

Okay, but I was asking about installing the new os in the future. Let's say I have created an ATV os and four PS "partitions". ATV is bottled and configured with additional apps. I want to replace one of the PS partition with the PCP - should I just check the proper partition and replace it with the PCP image (or any other image) without loosing my ATV system? What about the second, third and so PS "partitions"?

procount, 2022-01-31:
Yes. (sorry I misunderstood what you were asking).
Replacing or Reinstalling an OS does not affect the other OSes that are already installed. This is an improvement over NOOBS.

godzinilla, 2022-01-31:
> Yes. (sorry I misunderstood what you were asking).
> Replacing or Reinstalling an OS does not affect the other OSes that are already installed. This is an improvement over NOOBS.

No need to sorry. English is not my main language so I might wrote something not in the correct way :)

Thank you for all responses, I really appreciate your help. I will check all of this tomorrow and go back to you with the results, especially with this warning-thing with pcp.

godzinilla, 2022-02-01:
So the `Checking for boot files on mmcblk0p1... Trying to mount /dev/mmcblk0p1. newcp.cfg not found on mmcblk0p1` message shows on the source sd card (without PINN) and on the PINN version to. So no issue here.

closed 2022-02-01 by godzinilla