procount/pinn #622: Doubt about OS Updates

MopheusDG, 2022-01-31:
I have a doubt, when running PINN the Select OS screen shows me that, for example, Raspberry Pi OS Full has a [Recommended][Update]. My question is, what is the difference between those updates, and booting into the OS and running an apt update, etc... ?

I know updating via PINN will loose all my OS data, is that right ?

Sorry for the noob question.
Thanks.

procount, 2022-01-31:
Yes, you are totally right.
It's just an indication that a new version has been released.

lurch, 2022-01-31:
Most of the time you can indeed just `apt update` from inside the existing OS image. It's only when e.g. upgrading [from Buster to Bullseye](https://www.raspberrypi.com/news/raspberry-pi-os-debian-bullseye/) that it's recommended to do a new installation.

MopheusDG, 2022-01-31:
Thanks both for ur answers. By the way, this does not apply to OSes like Lineage, right ? That one needs to be updated via Pinn. Correct ?

procount, 2022-01-31:
It depends on the OS.
If the OS has an upgrade facility, always use that.
PINN does not "upgrade" an OS; it can only replace it with a new version which will wipe any existing data in that OS.

MopheusDG, 2022-01-31:
Thanks @procount . I'll close the comment and hope this helps another noob like me.

closed 2022-01-31 by MopheusDG