procount/pinn #623: Raspberry Pi OS 64 bit (Desktop and Lite) out of BETA - request for an images

godzinilla, 2022-02-03:
Hello @procount !

Raspberry Foundation at 28th of January 2022 finally pulled the 64 bit images out of the beta program. Images for Desktop and Lite are available. So it is time to prepare these images for the PINN :)

https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-64-bit

lurch, 2022-02-03:
I _believe_ @XECDesign is currently working on the 64-bit Raspberry Pi OS images for NOOBS (?), and once he's done that they'll automatically appear in PINN as well.

XECDesign, 2022-02-03:
Should've already been in NOOBS yesterday.

lurch, 2022-02-03:
Oh cool! Are they not appearing in PINN @godzinilla ?

godzinilla, 2022-02-03:
> 

Images are available :) Sorry, did not checked it before creating an issue :O