procount/pinn #624: Replace PiOS with new OS

MopheusDG, 2022-02-03:
Hi, I'm using PINN v3.7.5 and I've installed many OSes without issues. Unfortunately, I installed Pi OS 32 bits and now that 64 bit is out of Beta I wanted to replaced but can't. Is it not possible to do so ?

If select the Pi OS 32 bits on the menu, and select REPLACE, I can only (as New OS) a list of my current installed OSes, not the FULL list of available on the web.

Can someone clarify if it's me or there's no other option than reinstalling ALL OSes and starting from scratch.

Thanks a lot.

godzinilla, 2022-02-03:
@MopheusDG you need to use download option on selected os. Download it to attached USB drive (other that your PINN drive; need (?) to be formatted as FAT32). Then after download you will be able to replace the OS. I have done this today, the same scenario: Lite 32 replaced with Lite 64.

procount, 2022-02-03:
@godzinilla - actually it is NOT necessary to download them first. @MopheusDG has a different issue.

Please see the instructions for this at https://github.com/procount/pinn/blob/master/README_PINN.md#replace-individual-oses .
The list of OS replacements you can choose from comes from the list of selected OSes on the Main Menu. Any installed OSes are ticked by default, which is why you see them in the drop down list. Simply tick the Pios 64-bit OS, then head to the Maintenance Menu and select the OSes you want to be replaced. For each of these OSes you can then select which OS you want to replace it with.

MopheusDG, 2022-02-04:
Hi @procount, I've tried the normal steps you refered but that doesn't work. I don't know why, it seems like it's considering my local store the only OS repository, so I only can replace with those I already have.
Maybe @godzinilla is right and downloading it to another drive could work, but I couldn't find the way to do it either, as I still get my local OSes only.
Tried connecting through Wifi just in case the LAN connection was an issue, doing a Reload Repository still get's me my installed OSes only, no new one's.

Could this be related I created the partitions using https://pinn.mjh.nz/ ?

procount, 2022-02-04:
Yes, that's definitely your issue. Sorry I didn't think to consider that. Using https://pinn.mjh.nz restricts your choice of OS to those you selected on Matt's website.
So to restore PINN to normality, on your PINN partition, just copy recovery.cmdline.new to recovery.cmdline and reboot it. Now you should be able to see the 64-bit OS and select it as a replacement.
It should look like:
```
quiet ramdisk_size=32768 root=/dev/ram0 init=/init vt.cur_default=1 elevator=deadline repo_list=http://raw.githubusercontent.com/procount/pinn-os/master/os/repo_list.json loglevel=2 sdhci.debug_quirks2=4
```

MopheusDG, 2022-02-04:
Thanks @procount that was the issue. Now replacing Pi OS perfectly fine !

Thanks for your support and for PINN.

closed 2022-02-04 by MopheusDG