procount/pinn #625: Typos in os_all.json

paulwelka, 2022-02-04:
I *think* there are a couple of typos in https://raw.githubusercontent.com/procount/pinn-os/master/os/os_all.json that prevent lineage17-rpi4 OS's from installing. Specifically...

The OS "lineage17-rpi4-atv" won't install because, in the json file...

"http://downloads.sourceforge.net/project/pinn/os/lineage17-rpi4-atv/system.tar.xz"
 SHOULD BE
"http://downloads.sourceforge.net/project/pinn/os/lineage17-rpi4-atv/system.img.xz"

 and

"http://downloads.sourceforge.net/project/pinn/os/lineage17-rpi4-atv/vendor.tar.xz"
  SHOULD BE
"http://downloads.sourceforge.net/project/pinn/os/lineage17-rpi4-atv/vendor.img.xz"

The OS "lineage17-rpi4" won't install because, in the json file...

"http://downloads.sourceforge.net/project/pinn/os/lineage17-rpi4/system.tar.xz"
  SHOULD BE
"http://downloads.sourceforge.net/project/pinn/os/lineage17-rpi4/system.img.xz"

and

"http://downloads.sourceforge.net/project/pinn/os/lineage17-rpi4/vendor.tar.xz"
  SHOULD BE
"http://downloads.sourceforge.net/project/pinn/os/lineage17-rpi4/vendor.img.xz"

I could be wrong, but I think that's why I'm seeing an "Error 404: Not Found" error when trying to install those particular OS's.

procount, 2022-02-04:
Ah yes. Sorry. That's the trouble with sharing scripts between versions :(

procount, 2022-02-04:
Should be sorted now.

paulwelka, 2022-02-05:
Looks like that did the trick. I just installed lineage17-rpi4-atv via PINN with no errors. Thanks.

closed 2022-02-09 by procount