procount/pinn #628: Duplicate titles in no known order

RanchoHam, 2022-02-18:
After making an os back-up, I attempted to test the back-up by installing it into a blank project space.  When I looked at the resulting Maintenance (or Boot) menu, I could not tell the difference between the original and the restored back-up.

I did a similar thing with installing another copy of Raspbian 32 for a test installation.  During the installation I changed the hostname to a new hostname.  However, when I went to destroy the test environment; I could not differentiate between the two installations, chose the second in sequence, and subsequently wiped the non-test environment.

My question is how can I change the label of the OS in the Maintenance and Boot menus?

procount, 2022-02-18:
You can rename one of the OSes using the Alias feature. I think it is on the Maintenance menu.

RanchoHam, 2022-03-04:
OK, I finally had a moment to exhaustively test the backup restoration process.  

There is indeed an Alias button in the Maintenance menu, and it does what needs to be done, but only if used wisely.  
1.   back-up an OS
2.   boot the original OS and make a file in the root (/) directory by executing "sudo touch $HOSTNAME-orig"
3.   restore the back-up to a project workspace
4.   alias one of the OSs as "OS back-up test"
5.   boot "OS back-up-test"
6.   verify that there is no "/$HOSTNAME-orig" original file.  If the file exists, then change the alias to "OS original".
7.   continue to test the back-up to satisfy it works

So my recommendation is for PINN to test after a back-up restore operation for an identical OS name and offer to alias the newly restored OS before leaving the back-up procedure.

procount, 2022-03-07:
Alternatively, when you create a backup, you don't have to use the suggested date format that is appended to the OS name - you can change this to something else. Additionally you can edit the description field to describe exactly what this backup version is.

RanchoHam, 2022-03-07:
The issue is not with the date or description fields, those show up in the restoration process just fine.  During the backup process, the OS name field is NOT modifiable.  The problem is after the restoration to a different project space rather than overwriting the existing project space (which works just fine).  The behavior I am seeing is that the Alias seems to be applied to all projects with identical OS names.  (Either that or the Alias is being stored with the backup.)

paulwratt, 2022-03-18:
this looks good, it will help the "o" comes before "r" too

looking forward to seeing/hearing other peoples test results

closed 2024-07-17 by RanchoHam