procount/pinn #629: Black Screen after Recalbox 8.0.2-Electron Update

prodigy213, 2022-02-19:
My Device: Raspberry Pi 3 Ver. 2

My Actions: System (Recalbox) performs automated reboot after Recalbox 8.0.2-Electron update. Wait for "OFFLINE INSTALL" to complete. System performs automated reboot again after offline install. Some text quickly flashes (sorry I couldn't read it) then HDMI "resets", Recalbox "boots" to black screen. Motherboard green light calms down (stops flashing) after ~5 minute wait. NIC green light continues to flash. Keyboard strokes, Alt+Ctrl+Del/Back does nothing. Unplug power [while MB green light is off]. Power on after 3-9 second wait.

The update message given states:
- Fix 4k resolution issues on RPi4
- We are sorry but we must rollback the "fix" of the black screen as it caused other issues.
- If you still have an issue with black screen after this upgrade, please follow the instructions on https://fix.recalbox.com

At the URL it states:

Known issues
Black screen issue
[🇫🇷 Version Française](https://wiki.recalbox.com/fr/recalbox-8-0-faq#ecran-noir-sur-rpi3-ou-rpi4)

If you have a black screen issue on RPi4 or RPi3, you can fix it following those simple steps:

-Put your SD card in your PC
-In the RECALBOX partition, open the file cmdline.txt
-Add this text at the end of the line : video=HDMI-A-1:D video=HDMI-A-2:D
-The file should look like this:  dwc_otg.fiq_fix_enable=1 sdhci-bcm2708.sync_after_dma=0 dwc_otg.lpm_enable=0 console=tty3 consoleblank=0 loglevel=3 elevator=deadline vt.global_cursor_default=0 logo.nologo splash label=RECALBOX rootwait fastboot noswap video=HDMI-A-1:D video=HDMI-A-2:D
-Save the file
-Eject the SD card properly
-Test again in the Raspberry Pi.

I have done all of thee above fix recommended by recalbox but I still get a black screen. In the PINN menu, I selected "Fix" under the Maintenance tab. I checked all of the boxes, it gave errors on all except for "Check Permissions".

-File system check at the end gave me "recalbox-rpi3 filesystem: error detected  DONE-some errors detected"
-Re-run partition_setup.sh at the end gave me "recalbox-rpi3: Error executing partition setup script   recalbox-rpi3 paritions_setup.sh script error   DONE - some script errors detected"
-Show Debug Log said this at the end, "sed: /mnt2/cmdline.txt: No such file or directory" on the "Show Log" display.

If I perform a "Reinstall" from the maintenance tab, I am able to use verison 7.1.1-Reloaded.

procount, 2022-02-19:
So I can try and replicate this, what did you have installed initially - 7.1.1-Reloaded installed via PINN?
There were 2 recalbox versions - the official one, and one I modified with "USE THIS ONE" in the description. Which one did you have installed?
Did your Recalbox 7.1.1 upgrade automatically to 8.0.2, or did you have to download the file yourself and do an Offline Install?

There is a new official version of Recalbox 8.0.2 now available in PINN. It is not perfect and I am in the process of improving the meta-files so that it works properly.

prodigy213, 2022-02-21:
@procount Thanks for all your hard work. I hope I'm not adding on to you list issues to resolve. Thanks to how PINN is setup it's easy to reinstall/restore the OSes.

Yes, I did use the one installed initially on PINN. I installed "USE THIS ONE" because in the past I have installed the other Debian OS not labeled "RECOMMENDED" (i believe it's labeled that) and I had some issues with it. Thus I try to be aware of the labels posted.

I had performed the Upgrade myself. Whenever Recalbox would load it would have that popup stating it past version will not be supported. I will try the new one available on PINN in a few hours.

procount, 2022-02-21:
I've never tried to upgrade Recalbox before - I just don't have the time to test everything. I have got Recalbox pi4 installing; pi3 versions should also work. However, I have identified 7 issues with the recalbox metafiles, so I will address these shortly and try to get the maintainers to fix them.

prodigy213, 2022-02-21:
@procount I would like to be as helpful as I can but IDK how PINN,NOOBS and the like work. IDK what metafiles do nor where are they stored. Would you happen to know of a manual or document I can read?

Whenever you resolve the metafile issues please let me know. Ill be more than happy to test.

procount, 2022-02-21:
Don't worry about them - just something I need to sort out.

prodigy213, 2022-02-21:
Roger that. Whenever the metafile issues are resolved I'll be ready.

procount, 2022-02-21:
ok will let you know, but you should be able to use it now.

prodigy213, 2022-02-21:
@procount I just performed a compete install and the version that showed up on the RPi3 Ver. 2 was the 7.1.1-Reloaded. I didn't download and installed PINN; starting off fresh. I updated PINN instead and then performed the installation of all OSes I wanted all over again.

procount, 2022-02-25:
There is a new version of PINN available with mods for Recalbox's idiosynchrasies. Please use v3.8.1
I have also revamped all the metafiles for Recalbox v8 and removed Recalbox 7.1.1

prodigy213, 2022-03-02:
I have updated PINN to v3.8.1 Built Feb 23 2022 and I installed Recalbox onto SDcard alone and it said that some OSes installed with some errors. How do I access that error message?

procount, 2022-03-02:
Typically that error occurs if you have a command shell open which prevents PINN from modifying the partition table.  You could try re-installing immediately after rebooting PINN. I couldn't test the RPI3 version specifically because my 3B had gone missing at the time. I will search for it and retest it.

procount, 2022-03-02:
Found the problem. Reboot PINN TWICE and reinstall Recalbox for RPi3. It should then work.

prodigy213, 2022-03-03:
I was forced to perform a clean install today using the latest version of PINN lite. Since my SDcard is close to 128GB I performed a multiple OS installation. During the installation for Recalbox, I received the message, "Error: OS: 'Recalbox_-_Pi2' needs a partition label of 'SHARE' which is not available.

procount, 2022-03-03:
The Recalbox maintainers have gone down the route of referencing their partitions using labels, which gets around the problem of installing it on USB or SD card, but because the labels are hardcoded they need to be unique. This prevents installing 2 copies of Recalbox on the same system, or with any other OS that has the same labelled partition. So PINN prevents installation if the required partition labels are already used. I have already mentioned this to them. Using PARTUUIDs would have been better.

prodigy213, 2022-03-04:
@procount Thanks for the update. Would you happen to have an ETA on when they will fix the issue? I agree, they should be choosing the option that is most flexible/compatible with multiple operating systems. Do I have a way of renaming the partition during installation or post install?

procount, 2022-03-04:
I can only point out the issues to them. It's up to them if and when they fix them.
https://gitlab.com/recalbox/recalbox/-/issues/2186
You can rename the boot partition, but the other 2 are fixed, unless you want to change their init script. Using dev format may work, except if the boot partition number ends in an 8 or 9.

prodigy213, 2022-03-12:
@procount I had a little time on my hands and I was able to install Recalbox 8.0.2. as a standalone (using dd to copy the img file from recalbox.com to the sdcard). There's so many issues with 8.0.2 I had rolled back to Recalbox 7.1.1. by adding it to the PINN os directory/folder. Would it be possible to have both versions available for online installation?

procount, 2022-03-12:
OK, I will restore the old version shortly with a new name so they can co-exist.

prodigy213, 2022-03-13:
@procount Thanks so much! I haven't had the time to perform QC on the modifications you've made. I should have some time to check today and will close this ticket if everything goes smoothly. 

I know you're busy with bigger things. Without having you do too much work, and please let me know if I can help, how hard is it to get it upgrade from Recalbox 7.1.1 Reloaded to Recalbox 7.2.2 Reloaded? I'm using the file(s) provided at https://archive.org/details/recalbox-x86_64.img

I would recommend that anyone with a Rpi3 to NOT upgrade to Recalbox 8.0.2 at the moment. Issues like this needs to be resolved first https://gitlab.com/recalbox/recalbox/-/issues/2192. I don't get how they not noticed issues like that in the first round of QC's.

procount, 2022-03-15:
Fortunately I had copies of the 7.1.1 tarball versions, so I've been able to restore them. I have renamed the previous meta files to `recalbox7-rpiX` to distinguish them. They could be improved similar to the latest Electron versions, but they should do for now. 
I have tested the installation of all of them, but only tested booting the rpi4 version. I assume the rest will still boot as before.

prodigy213, 2022-03-17:
@procount Thanks for everything! Everything works as it did before. Do you need any assistance with upgrading 7.1.1. to 7.2.2?

closed 2022-03-17 by prodigy213

procount, 2022-03-17:
I've downloaded the 7.2.2 images so I'll convert them and probably replace the 7.1.1 files with them when I find some time.