procount/pinn #630: Is it possible to boot to next device in line?

NumberOneGit, 2022-02-23:


procount, 2022-02-23:
?

NumberOneGit, 2022-02-23:
I hit the submit button by accident

NumberOneGit, 2022-02-23:
Would this solution work?  I could use pinn.mjh.nz and do a normal partition layout with a dummy OS or project space that takes up the rest of remaining space on my USB drive, then install one OS on the SD also with PINN.  If I copy over the boot folder from the SD card to the boot folder of the "dummy OS" (while properly altering the PINN files in that folder), I should be able to edit cmdline.txt to point to the other drive as root filesystem.  Then I could format the ext4 of the "dummy OS" to use as a large data partition which is the way my drive is currently setup.  I'm going to test this out, but it seems like it should work.

procount, 2022-02-23:
Basically, no. (in answer to posts 1&2)
The Boot drive is determined by the firmware, and in later Pi4s the firmware can use the EEPROM bootorder to select the order. PINN cannot influence that selection. Once PINN has selected an OS to boot, it reboots the Pi to the partition number on the same boot drive. It cannot boot an OS on another drive.
3) has been possible for a long while, and is the only way for devices prior to the Rpi3B to run their OS from a USB device. 
The boot partitions must always be stored on the same device as PINN, but the rootfs partitions can all be stored on the same device or a secondary device.
But you can't store _some_ OS rootfs on the main device and _some_ on the secondary device, due to the way PINN is designed to fill up a whole drive with the OSes. The only way to do that is to install all OSes to the main drive and then copy some rootfs from the main to the secondary drive and manually fix things up.

lurch, 2022-02-24:
@NumberOneGit I'm not sure what it is you're trying to do, but if you've not already seen it you might find the info at https://github.com/raspberrypi/noobs/wiki/Standalone-partitioning-explained and https://github.com/raspberrypi/noobs/wiki/NOOBS-partitioning-explained useful.

paulwratt, 2022-02-24:
@NumberOneGit can't you get around the issue by having the seperate OS installed on seperate PINN devices?

Or are you wanting to select any other devices partition from a single OS list/menu - What happens when that PINN device doesn't boot first?

NumberOneGit, 2022-02-24:
The setup I want is on one drive, OS1, OS2, then LineageOS.  On this particular project I am using the NVME/eMMC setup on my CM4 but would also want to do something similar with my 4B later on.   I installed the three OS setup above on my PINN NVME setup and I installed a vanilla Lineage to the eMMC.  I then took the four Lineage partitions from the NVME and used dd to overwrite the eMMC Lineage partitions, just using the vanilla Lineage to get a proper partition layout.  
I edited the os_config.json in the Lineage PINN folder to reflect the NVME for boot partition one and used the UUID for the other three partitions on the eMMC.   It boot once and worked as expected, but then would not boot again.  I'm still trying to read through the documentation to figure out what I'm missing but I haven't found anything on having root on a separate drive yet.

NumberOneGit, 2022-02-24:
@procount  Can I just have a regular installation of Lineage on the eMMC that will boot on it's own if the PINN drive isn't attached and create a single boot partition in PINN (which is first in boot order) that allows me to still be able to boot into that OS from PINN?  I know that if this is possible it would require a lot of manual editing, I'm just asking if there is any reason this won't be possible.

procount, 2022-02-24:
Android/Lineage is way more complicated to multiboot than any other OS. You can't just copy the partitions. os_config.json is only used to rebuild installed_os.json if the settings partition gets deleted, so modifying that will have no effect.

> Can I just have a regular installation of Lineage on the eMMC that will boot on it's own if the PINN drive isn't attached and create a single boot partition in PINN (which is first in boot order) [...]

PINN can only boot an OS if its boot partition is on the same drive as PINN. It cannot boot an OS on another drive, although the root partition can be on another drive.
The only way to do this would be to copy the boot partition onto the PINN drive so that there are 2 versions of it and do any fixups manually. I wouldn't recommend this, especially with something like Lineage.

NumberOneGit, 2022-02-24:
In cases where I would need TWRP I would remove the device from wherever it was housed.  Most system related needs I can do over wired adb.  It sounds like you are describing something similar to what I am..  cloning the boot folder into PINN and manually editing fixups.  I figured it would be quite a challenge as I know Lineage and other Android based OS's are very complex and I knew I may not get it to work, but it is the primary use of the device.  I feel like this goal is more achievable (if it works) than trying to figure out whether it's possible to boot Lineage from pcie.  It will at least be a good way to learn the NOOBS/PINN inner-workings even if it fails.

paulwratt, 2022-02-24:
> cloning the boot folder into PINN and manually editing fixups.
it should work if your `cmdline.txt` sets the root device with a UUID (as opposed to a PARTID) - how do you intend to handle the remount of `/boot` (which will be different for different PIN boot devices), as the _both_ need to be sync'ed.

lurch, 2022-02-25:
> It will at least be a good way to learn the NOOBS/PINN inner-workings even if it fails.

It might be easier to experiment with setting up Raspberry Pi OS to boot the way you describe, as that has a more "traditional" partition layout :wink:  And then once you've got to grips with how that all works, **then** try experimenting doing the same thing with Lineage?
Sometime it's easier to do baby steps, than it is to try diving straight into the deep-end... :slightly_smiling_face:

NumberOneGit, 2022-02-25:
@lurch Agreed, that was actually what I've been working on over the last few hours.  Thanks for the advice.

NumberOneGit, 2022-02-25:
@paulwratt   "How do you intend to handle the remount of /boot (which will be different for different PIN boot devices), as they both need to be sync'ed."  Could you go into a little more detail on this so I can do some research?  I also want to make sure I understand the question correctly.

paulwratt, 2022-02-25:
> Could you go into a little more detail on this so I can do some research?

(You need to test this to know exactly how the `/boot` partition handled, _before and after_ the OS is running, knowing that various OS handle this differenlty - some explicitly define it, some inherit it from the boot process) 

Depending on which device boots to the "same OS" then the `boot` folder will be different partitions. Because you have 2 seperate boot partitions on 2 different devices

If '/boot' is defined (say) in `/etc/fstab` then that is the only `/boot` partition that will get updated with any firmware or kernel updates provided by the package manager.

So you need to know which partition is mounted on `/boot` _after_ the OS boots, and mount the other partition somewhere and (for example) `rsync` it when it changes, _AND_ it needs to "fail gracefully" if that other partition cant be found.

(However) If you are sure you are not going to have anything in `/boot` updated, then all of this is irrelavant.

NumberOneGit, 2022-03-03:
@paulwratt Thanks for the info.  Thankfully, I am moving on from the Lineage option and will probably figure out a similar setup with either Ubuntu or RPOS on the eMMC.   As it turns out, re-compiling the Lineage kernel with the NVME boot flag config allowed all functionality to work OOTB, including TWRP.  I'm still continuing with the project in this thread but with a less complicated OS.