procount/pinn #631: Cannot Maintenance->Backup, and cannot Replace an OS; also no USB devices in sub menus.

al-jimenez, 2022-03-05:
I successfully created a PINN setup with 5 OS; but when I try to Backup or Replace or anything in any of the menu's and the USB dropdown items disappear;  also Backup is greyed out even after selecting an OS to backup; 

Pi 4 Model B (2019) Rev 1.2 (c03112) - 4GB.

Any help would be appreciated.

procount, 2022-03-05:
Backup - Best to use a FAT32 formatted USB drive. Make sure is has a folder in its root directory called `os`. This is where your backups will be stored. Without this folder, the USB drive will not be recognised as a backup device. Nad if youdon't have a recognised device, the backup icon will grey out. 

Replace - You must first select any OS that you want installed in the Main Menu. Then go to the Maintenance menu and select the OS you want to be replaced. Now you should be able to click the Replace button and select which new OS will replace your old selected OSes.

procount, 2022-03-17:
Any progress? Please close if the above has solved your issue, or provide more information.

al-jimenez, 2022-03-18:
No progress.  I recreated a new PINN but had the same issue.  I did/am using a FAT32 drive; I have used more than one USB drive;  I am on a Raspberry Pi 4B with 4GB, connected both USB drives to USB3 ports directly, and had an OS folder;  but PINN Maintenance menu does not see any USB devices at all;  I select REPLACE and it shows the existing already installed OS'es to choose from not a USB sourced list.   The USB devices only show up in the main menu and then do not allow browsing or reviewing them it only allows formatting them with PINN already selected OS's.    I have re-installed PINN with my select OS's and have moved on to other things.  I have tried many things on the assumption that my config is flawed but have run out of things to try.   If you want me to close this I can, but I feel the issue is still open.  Has anyone else reported this issue?  If not then I will close on the assumption that my Pi or other configuration is the issue.

procount, 2022-03-18:
>  I select REPLACE and it shows the existing already installed OS'es to choose from not a USB sourced list.

This function works backwards to what you are expecting. You first select the OS you want removed, THEN you select what you want it replaced with.

Since your USB drive is not showing up in the destination drive, I wonder what "USB sourced list" you have on your USB drive. PINN will not install standard .img files, so if you have a list of those on your USB drive, then I'm afraid they won't install. PINN installs are formatted differently than .img files to separate the content from the partition information (since the partition layout needs to be altered for a multi-boot environment).  There are details on my Wiki and in the README_PINN.md file to explain how to convert .img files to the .tar.xz format that PINN expects.

The main way to get OSes onto your USB drive for local installation is to use PINN's download function to download one of the may OSes that PINN supports onto your USB drive. But your drive first needs to be recognised as a backup drive by having the /os folder on it as mentioned in the previous post.

procount, 2022-03-18:
One more thing....
You mentioned plugging BOTH USB drives into the USB3 ports at the same time.
Are you using one of them as your PINN boot drive and installing OSes to it, or are you installing to an SD card?
If you are USB booting PINN then that's fine. If you were installing to an SD card, then having 2 USB drives with /os folders on them may confuse PINN; it only expects to have one additional USB drive with an /os folder on it.

al-jimenez, 2022-03-20:
I will try to give an answer to each of your responses.   I am really impressed and appreciate your detailed response. I had moved on from this issue as I found a workaround that allowed me to continue using PINN.   Anyway, in appreciation of your help, I agree I had a miss understanding as to how the features worked.   

Yes, I was definitively using FAT32. I use a MacBook but specifically went to my Windows machine to format them for FAT32.  I have a USB 3.0 thumb drive that is 32GB and I have a NVMe 250GB USB drive which I specifically formatted as FAT32 and created a folder name "os" in lower case.   My intent was to create image files (of backup instances of other Pi installs I had) in tar format for PINN to use as one of the multi-boot OS's.   I did used your description details on the Wiki README_PINN.md to convert the .img files to the .tar.xz format.

So I went to "Replace" selected the OS I wanted to replace, in my case it was "Ubuntu" selection that I had already installed and booted.   My intent was to get the target OS from one of the two USB from the "os" folder.  For example I had an instance of "Home Assistant" that I had already configured and backed up and created into a .tar file.  I wanted to replace "Ubuntu" with "Home Assistant".   When I selected "Ubuntu" to replace it, I got back only the list of OS that I had originally chosen when I first created the PINN boot. 

With regard to the "BACKUP function": I tried both USB drives, both formatted to FAT32 and with lower case "os" folder to backup my existing "Raspberry Pi 64" install.   When I selected Backup the USB selection was blank.  My USB did show up on the original menu as a destination for PINN install (and asked to format it), but did not show up from the Backup menu.   When I selected "BACKUP function" since it did not see the USB drive I did not get further as the "BACKUP function" was grayed out.

My original list of OS's in PINN was 6 OS:  Raspberry Pi OS 64, Raspberry Pi OS 32, Ubuntu 2110, Ubuntu 2004, RetroPi, and two Project Spaces.   I was attempting to replace the Ubuntu 2004 with my "Home Assistant" backup.

NOTE: I am booting the original PINN from a 500GB SSD.  I created it from the PINN website and selected USB (size=500GB) rather then SD card.    I have not tried to use the SSD as the location of my "os" folder.  I will have to try that.   My use of the USB drives:  I had each of the USB drives plugged in individually first.  I tried each of them individually, that tried the together.  They were recognized as targets for a new PINN install but not as backup or replace targets. 

I will try to create the /os folder on the SSD that I boot PINN from and see if it is recognized.  But as I mentioned the "BACKUP function" is grayed out and the "REPLACE function" only provides replacements from the existing installed OS's.

I really do appreciate your time & detail in responding back to me.  I have experience in software dev and I understand how complex this can get especially in trying to duplicate a user's environment and actions, even with the simplest software.   I also understand that I may not totally comprehend how to use PINN.   BTW: fantastic job you guys have done with this. It is so useful and beneficial to me and I continue using it.  Congrats on a job well done.

ArashiCZ, 2022-07-25:
Hello.
I have a simmilar issue. I'm using CM4 on a Waveshare baseboard.

- PINN is installed on a emmc on the CM4 - 16GB.
- My target drive for OS'es is a NVME SSD drive on the bottom of the baseboard - 256GB. 

I managed to install Raspi OS 64bit on the nvme ssd and it works perfectly. ( I ran benchmarks, stressberry etc.)

I'm struggling to get the backup function to work.
It is a development platform for me, so I would like to make several, regular backups on the ssd.
Button is grayed out.

Here is an image of the drives details:
[image]

Can I, please, get some guidance, how to properly backup os on the ssd?
Either 
- on the same ssda
- an external drive
- maybe a SMB in the future?

Thank you, very much.
PINN is an excelent platform that helps lot of us.

procount, 2022-07-25:
I don't have a CM4, so my help may be limited :wink:

>Can I, please, get some guidance, how to properly backup the installed os either
> - on the same ssd
> - an external drive
> - maybe a SMB in the future?

same ssd is not possible - it needs to be a different drive.
By SMB you mean a samba share, or remote network file system of some sort - I will have to think about that one.
So for now your only option is an external drive (e.g. USB memory stick).

After googling some waveshare baseboards, I assume you have a USB port on there? Most seem to have at least one.
I would suggest using a 32GB usb stick and format it as FAT32. (Any bigger and you risk it being formatted as exFAT which I don't think is supported for this operation.)
Create a folder on it called os.
Boot your CM4 into PINN.
Click on the "more" button to move to the Archival menu.
Does the Destination Drive drop down have any drives listed in it? - which?
Plug your USB stick in.
Does the USB drive show up in the Destination Drive? - probably as "sda:"?
Click on the "more" button to move to the Maintenance menu.
Does the USB drive still show up in the Destination Drive? - probably as "sda:"? - Make sure to select it.
You should see a list of your installed OSes - Raspios 64
Click the checkbox next to it. The Backup up button should become coloured/enabled.
Click the backup button.
If PINN complains about the OS being unsupported and needs an update, make sure you are connected to the internet.

ArashiCZ, 2022-07-27:
Hello @procount.

Thanks again for such a quick response.
Today I bought an USB 3 flash drive just for executing the backups.

Device successfuly listed in running RaspiOs 64bit.

- Drive details:
[image]´

- Drive contents:
[image]

However, my usb drive is not listed in the PINN menu, neither of them.
Here are some images:
[image]
[image]

Can you, @procount , help me, please?

Thank you, very much!

procount, 2022-07-27:
What size is your new usb3 drive?
Have you formatted it as FAT32?
You must create a folder called os on it, otherwise it will not be listed.

ArashiCZ, 2022-07-27:
Hi!

Asi i posted the pictures above, you can see that the drive is 16GB FAT32 with a folder only one folder, called os.
Still doesn't get listed.

Thank you.

ArashiCZ, 2022-07-27:
No need to be sorry, I'm to blame.
I figured, that you had no time. I didn't compose the comment clear enough thou.
Really appreciate that you took your time and trying to help me with this.

- Output of /tmp/debug
```
3.8.3
Board revision is  11546945
QVNCServer created on port 5900
mount: wrong fs type, bad option, bad superblock on /dev/mmcblk0boot0,
       missing codepage or helper program, or other error

       In some cases useful info is found in syslog - try
       dmesg | tail or so.
mount: wrong fs type, bad option, bad superblock on /dev/mmcblk0boot1,
       missing codepage or helper program, or other error

       In some cases useful info is found in syslog - try
       dmesg | tail or so.
Found recovery.rfs at "mmcblk0p1"
PINN drive: "/dev/mmcblk0"
mount: /dev/mmcblk0 is already mounted or /mnt busy
Simulate OK
CecListener thread started
Looking for  "/mnt/cec_keys.json"
```

- Output of ls /dev/*
```
/dev/btrfs-control    /dev/ram2             /dev/tty43
/dev/cachefiles       /dev/ram3             /dev/tty44
/dev/console          /dev/ram4             /dev/tty45
/dev/cpu_dma_latency  /dev/ram5             /dev/tty46
/dev/fb0              /dev/ram6             /dev/tty47
/dev/fb1              /dev/ram7             /dev/tty48
/dev/full             /dev/ram8             /dev/tty49
/dev/fuse             /dev/ram9             /dev/tty5
/dev/gpiochip0        /dev/random           /dev/tty50
/dev/gpiochip1        /dev/tty              /dev/tty51
/dev/gpiomem          /dev/tty0             /dev/tty52
/dev/kmsg             /dev/tty1             /dev/tty53
/dev/log              /dev/tty10            /dev/tty54
/dev/loop-control     /dev/tty11            /dev/tty55
/dev/loop0            /dev/tty12            /dev/tty56
/dev/loop1            /dev/tty13            /dev/tty57
/dev/loop2            /dev/tty14            /dev/tty58
/dev/loop3            /dev/tty15            /dev/tty59
/dev/loop4            /dev/tty16            /dev/tty6
/dev/loop5            /dev/tty17            /dev/tty60
/dev/loop6            /dev/tty18            /dev/tty61
/dev/loop7            /dev/tty19            /dev/tty62
/dev/mem              /dev/tty2             /dev/tty63
/dev/mmcblk0          /dev/tty20            /dev/tty7
/dev/mmcblk0boot0     /dev/tty21            /dev/tty8
/dev/mmcblk0boot1     /dev/tty22            /dev/tty9
/dev/mmcblk0p1        /dev/tty23            /dev/ttyAMA0
/dev/mmcblk0p2        /dev/tty24            /dev/uinput
/dev/mmcblk0p5        /dev/tty25            /dev/urandom
/dev/mmcblk0p6        /dev/tty26            /dev/vc-mem
/dev/mmcblk0rpmb      /dev/tty27            /dev/vchiq
/dev/null             /dev/tty28            /dev/vcs
/dev/nvme0            /dev/tty29            /dev/vcs1
/dev/nvme0n1          /dev/tty3             /dev/vcs2
/dev/nvme0n1p1        /dev/tty30            /dev/vcs3
/dev/nvme0n1p2        /dev/tty31            /dev/vcsa
/dev/nvme0n1p5        /dev/tty32            /dev/vcsa1
/dev/nvme0n1p6        /dev/tty33            /dev/vcsa2
/dev/port             /dev/tty34            /dev/vcsa3
/dev/ptmx             /dev/tty35            /dev/vcsu
/dev/ram0             /dev/tty36            /dev/vcsu1
[... 301 lines cut ...]
[   27.107810] random: 1 urandom warning(s) missed due to ratelimiting
[   32.486070] cam-dummy-reg: disabling
[   32.486087] cam1-reg: disabling
```

Drive doesn't get reckognized, atleast thats is what i think.

- Output of a lsblk
```
NAME         MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
mmcblk0boot0 179:32   0     4M  1 disk
nvme0n1      259:0    0 238.5G  0 disk
|-nvme0n1p5  259:3    0    32M  0 part
|-nvme0n1p1  259:1    0    63M  0 part
|-nvme0n1p6  259:4    0 238.4G  0 part
`-nvme0n1p2  259:2    0     1K  0 part
mmcblk0boot1 179:64   0     4M  1 disk
mmcblk0      179:0    0  14.6G  0 disk
|-mmcblk0p2  179:2    0     1K  0 part
|-mmcblk0p5  179:5    0    32M  0 part /settings
|-mmcblk0p1  179:1    0    63M  0 part /mnt
`-mmcblk0p6  179:6    0   256M  0 part
```
Here I only see the EMMC module, where then PINN is installed and the SSD, where the
OS gets stored and ran from.

procount, 2022-07-27:
So as PINN is not seeing your USB drive at all, I'm guessing that your Waveshare baseboard is using a different USB chip which PINN (being a minimal system) does not have the correct USB driver for. 
If you can identify the driver that is used for it in RaspberryPiOS (or the chip) then I can see if I can add it to PINN.
Try using `lsmod` and `lsusb`

procount, 2022-07-27:
Try adding this to PINN's config.txt
```
 dtoverlay=dwc2,dr_mode=host
```

ArashiCZ, 2022-07-28:
First of all, thank you so much, for taking your time on this.

Got it working. I totaly ommited that the usb ports are by defualt turned off.
There is a note by the manufacturer to modify the config.

- Note by the board manufacturer
```
  9: USB2.0 is closed by default, if you need to open it, you need to add dtoverlay=dwc2,dr_mode=host in config.txt.
```

- Image of the PINN

  [image]

- Successful backup
[image]

Thank you, again. PINN is awesmome.

procount, 2022-07-28:
Yay!
I saw that note which prompted my last post.
Glad you got it working

closed 2022-07-28 by procount