procount/pinn #634: "Pico-ized" RISCOS with PINN doesn't see it's partition

mbodomi, 2022-03-09:
I installed RISCOS with PINN and removed the filecore partition and made it boot to BASIC (theres info on the forum)
RISCOS Pico was a BASIC 50 years celebratory edititon, wich booted into BASIC interpreter.
RISCOS sees the recovery partition, not RI_OS_BOOT.

procount, 2022-03-09:
I found this post -> https://www.riscosopen.org/wiki/documentation/show/Software%20information:%20RaspberryPi:%20RISC%20OS%20Pico%20RC5
which directed me to the downloads page -> https://www.riscosopen.org/content/downloads/raspberry-pi but I could not find the pico version there.

But in any case, RiscOS uses a strange file system and requires a special hybrid binary blob to be installed at a specific location of the SD card that is specific to the version of RiscOS. So it's probably not surprising that the Pico version does not work and I'm afraid I have no idea how to make it work. You may need to get the RiscOS devs to create another NOOBS binary blob for this version.

lurch, 2022-03-10:
> You may need to get the RiscOS devs to create another NOOBS binary blob for this version.

Yeah, from what little I remember (IIRC RISCOS couldn't read MBR partition tables, and so instead hardcoded it's partition to a fixed offset), that's likely to be the solution. And you'd (probably) also have the caveat that you'd only be able to install one RISCOS or the other in PINN, not both at once.

paulwratt, 2022-03-16:
Also note that the original RPi4 build os RISCOS uses HDMI1 (the one closest to speaker/composite jack) _not_ HDMI0 (the one closest to USB-C power)