procount/pinn #635: Secure remote access

procount, 2022-03-10:
I have added the ability to disable dropbear SSH passwords and use keys instead for secure remote access.
I haven't found a way to add a password to the Qt VNC protocol, but having created a secure SSH connection to PINN, it should be possible to create an SSH tunnel to the VNC port.
Whilst I can create a tunnel from putty via PINN to _another_ device, I can't seem to tunnel to a port on the same PINN device for VNC. 
When I try to connect to VNC through the tunnel, putty claims the port was closed by the remote port. 

Anyone got any idea what the issue could be?
I'm not sure whether it could be dropbear or Qt's VNC that is the problem. I could try replacing dropbear with sshd, but I imagine it would take up more space. I'm sure this works fine with sshd in Raspbian.

cc: @lurch, @maxnet, @XECDesign

maxnet, 2022-03-10:
ifconfig does show lo interface is up?

procount, 2022-03-10:
Ha! Actually NO. Lo does not exist. How should I enable it?

procount, 2022-03-10:
```root@recovery:~# ifconfig
eth0      Link encap:Ethernet  HWaddr DC:A6:32:90:A8:3D
          inet addr:192.168.2.75  Bcast:192.168.2.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:32439 errors:0 dropped:28 overruns:0 frame:0
          TX packets:3203 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:5170917 (4.9 MiB)  TX bytes:2176819 (2.0 MiB)

eth1      Link encap:Ethernet  HWaddr 00:E0:4C:36:09:B7
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

wlan0     Link encap:Ethernet  HWaddr DC:A6:32:90:A8:3E
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)
```

maxnet, 2022-03-10:
>Ha! Actually NO. Lo does not exist. How should I enable it?

If there is a /etc/network/interfaces inside the image, try adding "ifup -a" to the /init startup script.
Otherwise try something like "ifconfig lo up"

procount, 2022-03-10:
I tried `ifup lo` and that seemed to make it appear too.

procount, 2022-03-10:
And now it works!
Many thanks!

closed 2022-03-10 by procount