procount/pinn #637: CM4: NVMe boot fails with 'failed to open nvme

procount, 2022-03-16:
Continuing from https://github.com/raspberrypi/rpi-eeprom/issues/419 by @mchaker

> (Is there a way to avoid the "boot into PINN, then reboot into RaspiOSLite" loop and just directly boot into RaspiOSLite?)

Yes, add an autoboot.txt file to PINN's partition with the line boot_partition=6 in it (IIRC). Remove it to restore normal operation.

> I think the /boot partition gets made... on the eMMC, and the data partition gets made on the NVMe SSD?

Yes, the /boot partition must always be on the same partition as PINN.

> It would be nice to be able to choose btrfs (with the SSD option/flag enabled) instead of ext4 for the root partition but I'm at least thankful it works now!!!

I'm sure I supported BTRFS in PINN to allow XBIAN to be installed. So if you crafted your own meta files for RaspiOSLite you could probably do it.

EDIT: Yes btrfs is supported.

> Thank you for the tips @procount ! What is a "meta file"? (a link to an example would be fine of course -- I'm guessing there's probably one in the PINN repo?)

Metafiles contain metadata, which is data that describes data.
So in the case of PINN, when you install an OS, PINN downloads a tarball for each partition from a website (the "data"), but it also downloads a host of JSON formatted files ("metafiles") that contain information about the OS and its tarballs (the "metadata") that informs PINN how to install it, how big it is, descriptions, version info etc.
You can read all about them in https://github.com/procount/pinn/blob/master/README_PINN.md and on my wiki.
Take a look at https://sourceforge.net/projects/pinn/files/os/ Each folder there contains the data and meta files for an OS. The partitions.json file indicates which file system to use for each partition. If you were to duplicate a set of files for local installation and change ext4 to btrfs, it _should_ work.

lurch, 2022-03-17:
> If you were to duplicate a set of files for local installation and change ext4 to btrfs, it should work.

...as long as the Raspberry Pi OS kernel has btrfs support compiled directly into the kernel, and not as just a module? (I've got no idea, I've never looked at btrfs)  If btrfs was only a kernel-module, you'd have a bit of a chicken-and-egg situation :hatching_chick:  :wink:

procount, 2022-03-17:
Yes good point, btrfs needs to be compiled in. (Hence why I put _should_ in italics - I've not tested this. :wink:. )

mchaker, 2022-03-19:
Thank you all for the details - once I get a chance (early April?) I'll test this out and report back. :)