procount/pinn #638: Raspberry Pi OS 64-bit Bullsey back-up not working

RanchoHam, 2022-03-16:
I used to be able to back-up RaspiOS Buster 64-bit instance, but the current Bullseye 64-bit installation cannot be backed up.

procount, 2022-03-16:
You should be able to back it up. What error messages are you getting? Any useful info in the debug log?

RanchoHam, 2022-03-17:
Error dialog reads:
"Warning:  Unsupported Operating System(s) detected.  PINN currently cannot backup the following OSs correctly:
Raspberry Pi OS (64-bit)"
The debug log does not look helpful to me but here it is anyway.  (BTW I had to rename with .txt to it for GitHub to accept it.)
[debug.txt](https://github.com/procount/pinn/files/8279639/debug.txt)

procount, 2022-03-17:
Whilst connected to the internet, reboot PINN twice, and then you should find Raspberry Pi OS (64-bit) can be backed up.

lurch, 2022-03-17:
"Have you tried turning if off and on and off and on again?" :rofl:

procount, 2022-03-17:
Just "to be sure, to be sure"! :wink:

RanchoHam, 2022-03-17:
> "Have you tried turning if off and on and off and on again?" 🤣

Multiple times on multiple machines after multiple software updates.

procount, 2022-03-17:
> Just "to be sure, to be sure"! 😉

It is St. Patrick's Day after all.

> Multiple times on multiple machines after multiple software updates.

The reason for rebooting twice (assuming PINN is already running) is that Raspberry Pi OS (64-bit) is missing some flags to say it could be backed up in os.json, so I have to override it, but that entry was missing. So on the first boot PINN will download the new overrides, but they are not effective until the next reboot.

In an offline situation where overrides cannot be updated, it may be necessary to add
```
"supports_backup" : true,
```
to "/settings/os/Raspberry_Pi_OS_(64-bit)/os.json"

closed 2022-03-17 by procount