procount/pinn #640: Compiling Issue

NumberOneGit, 2022-03-21:
I have been trying to compile PINN for a few days and have not been able to get past this hang-up...  I'm not familiar with Buildroot and any suggestions would be helpful.  I've mainly tried using uclibc-menuconfig to try to edit the path to try to point to the right files, but I haven't had much success.   

>>> linux dc6771425e9604650d1d57f7c69948be405f59a5 Building
PATH="/home/pi/pinn/buildroot/output/host/bin:/home/pi/pinn/buildroot/output/host/sbin:/home/pi/pinn/buildroot/output/host/usr/bin:/home/pi/pinn/buildroot/output/host/usr/sbin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin" /usr/bin/make -j5 HOSTCC="/home/pi/pinn/buildroot/output/host/usr/bin/ccache /usr/bin/gcc -O2 -I/home/pi/pinn/buildroot/output/host/usr/include -L/home/pi/pinn/buildroot/output/host/lib -L/home/pi/pinn/buildroot/output/host/usr/lib -Wl,-rpath,/home/pi/pinn/buildroot/output/host/usr/lib" ARCH=arm INSTALL_MOD_PATH=/home/pi/pinn/buildroot/output/target CROSS_COMPILE="/home/pi/pinn/buildroot/output/host/usr/bin/ccache /home/pi/pinn/buildroot/output/host/usr/bin/arm-buildroot-linux-uclibcgnueabi-" DEPMOD=/home/pi/pinn/buildroot/output/host/sbin/depmod -C /home/pi/pinn/buildroot/output/build/linux-dc6771425e9604650d1d57f7c69948be405f59a5 zImage
make[1]: Entering directory '/home/pi/pinn/buildroot/output/build/linux-dc6771425e9604650d1d57f7c69948be405f59a5'
  HOSTCC  scripts/extract-cert
.......
/usr/bin/ld: cannot find /lib/libc.so.0 inside /
/usr/bin/ld: cannot find /usr/lib/uclibc_nonshared.a inside /
/usr/bin/ld: cannot find /lib/ld-uClibc.so.0 inside /
collect2: error: ld returned 1 exit status
make[2]: *** [scripts/Makefile.host:95: scripts/extrac

/home/pi/pinn/buildroot/output/host/usr/arm-buildroot-linux-uclibcgnueabi/sysroot/lib/

procount, 2022-03-21:
Are you trying to compile this on a RPi?
Which OS are you compiling it on?
Have you installed all the pre-requisites?
Are you trying to build the latest version of PINN without modification?
You shouldn't have to modify uclibc-menuconfig.

NumberOneGit, 2022-03-21:
I'm trying to compile on RPOS Buster 32-bit because qt4-linguist-tools isn't available for Bullseye.  I didn't originally modify the uclibc-menuconfig and received this error, that was just the approach I took as it seemed that the paths needed modification.   The modification that I'm attempting is to use linux-menuconfig to enable USB gadget mode for access to PINN over SSH headless without internet access (USB_ETH=y, USB_ETH_RNDIS=y, USB_GADGET=y).

NumberOneGit, 2022-03-21:
I tried to build on Ubuntu 20.10 but was getting some strange errors that I didn't know where to start debugging.  I may try rolling back to 20.04 as I seem to be running into issues in general.  If I use linux-menuconfig to alter the kernel options, should I still use ./BUILDME.sh script or should I stick to just running the make command in the buildroot folder?

procount, 2022-03-21:
Start from a fresh download with all pre-requisites and just run ./BUILDME.sh.
That should work straight away, if not fix it until it does before modifying things.

There are three separate kernels that need to be built. Use the `./make-arch-linux-menuconfig.sh` script to modify their configuration. Pass arm6, arm7 or arm7l as a first argument to select which one to modify. Normally you should modify them all identically.

lurch, 2022-03-21:
Buildroot (and therefore PINN) is a pretty huge compile-job as it compiles **everything** from source, including all of the compilers / build-tools and all of Qt. So if you can, I'd definitely recommend trying to build on a PC / laptop running Ubuntu (even if inside a VM), rather than trying to build on a Raspberry Pi :wink:

procount, 2022-03-21:
I thought I would do a test to see if the latest changes for Ubuntu 20.04 still allow it to be compiled on a Pi4/4GB.
I'm using the Raspberry Pi OS Buster-Legacy 32-bit lite OS.
I copied all the download files first (you can do `make source` to download them all, but I often have to download the Linux or firmware files separately for some reason).
I'm building on a 16GB SD card, running in an Argon One case, but the fan is off, no overclocking. first build so no benefit of compiler caching. It's getting pretty hot! Been going for 2 hours and still only up to compiling Qt. We'll see how it goes.....

procount, 2022-03-21:
I got the same error.

NumberOneGit, 2022-03-22:
I switched over to a VM Ubuntu 20.04 on Windows.  It's still plugging along slowly..  I added libmpc-dev after receiving a missing gmp.h error and it seems to be about to finish (version with no changes)

NumberOneGit, 2022-03-22:
I completed the unaltered build successfully, ran the "./make-arch-linux-menuconfig.sh" script on each of the three architectures, then ran the "./BUILDME.sh" script.  I then copied over the output folder and booted up, but "zcat config.gz | grep CONFIG_USB_GADGET" shows "is not set".  Did I do this in the wrong manner?  Was I supposed to do something other than run the BUILDME.sh script after editing the menuconfigs using the script above or to run BUILDME with another arg?

procount, 2022-03-22:
It sounds like you did everything correctly.
The kernel gets rebuilt 3 times anyway so there should be no need for a make clean.

NumberOneGit, 2022-03-22:
Strangely, the kernelconfig-recovery files also pick up both the USB_GADGET and USB_ETH config settings but not the USB_ETH_RNDIS setting even though menuconfig showed all dependencies were met and it still shows up with a "y" in menuconfig when re-opening.

NumberOneGit, 2022-03-24:
It was just config.gz not updating that threw me off along with not realizing that the DWC2 kernel config was not set to dual role so the other changes weren't working.  All good now, thanks for your help!

closed 2022-03-26 by NumberOneGit