procount/pinn #643: Enhancement Request:  Add hardlink support to rsync-type backups

jharris1993, 2022-03-26:
Greetings everyone!

Sorry I haven't been around in a while, I've been recovering from illness and been busy.

Despite that, PINN has always been in the back of my mind and I am constantly looking for things that might improve things.

My apologies if this has already been asked - I didn't find it - but in the past there have been reports of "flakyness" with PINNified O/S's (especially when self-done) and/or restored backups.

One solution that was proposed was to enable extended attribute and ACL support with rsync's "X" and "A" flags, which I believe was eventually implemented.

One thing I recently discovered was the rsync "H" flag - "preserve hard links between instances of a file within a filesystem".  It normally defaults to "off" because it takes more memory and the "typical" rsync backup is a file-tree not an operating system - as in mirroring a repository.  Operating systems can, and do, use hard links to make a single instance of a file appear to be in more than one location without the problems associated with soft-links if a file is moved or replaced. (Soft links can become un-linked/broken under certain circumstances.)  Hard links also help support a seamless update process in ways that soft links cannot.

I did some experimenting with find, searching for hard links in a recent version of Raspbian/Raspberry Pi O/S and discovered that hundreds of files are hard linked.  Listing hard links to a file resulted in over 1600 lines of hard-linked files in Raspbian Legacy.

The result of rsync'ing without preserving hard links is that necessary system files may be referenced in more than one location, perhaps even using a different name - and if hard links are not preserved rsync copies them as separate files, breaking the hard links.  As a consequence, when these files are updated or a configuration is changed, only the original target file is modified.  This can result in various things becoming uncoupled and bizarre, un-reproducable behaviors from system to system.

Since the object of PINNifying an O/S image, (or creating a backup), is to create an identical - as if it were genetically cloned - instance of the operating system except for the necessary changes needed to PINNify the file.

I do not remember if the "H" option is already being used or not.  If it isn't may I humbly request it be considered.

Thanks!

procount, 2022-03-26:
PINN uses bsdtar not rsync. Does that make a difference?

jharris1993, 2022-03-27:
Procount,

 

If I remember correctly, the backup and restore process uses “plain vanilla” GNU tar, but the “PINNify” process uses BSD tar.

 

Please remind me why BSD tar is necessary for the initial PINNify process instead of GNU tar?

 

I will research the issue of preserving hard links in TAR.

 

Jim

procount, 2022-03-27:
Hard links are supported by all versions of tar, so this really is not an issue.

Bsdtar supports all the formats that plain vanilla tar does, but certain OSes required the additional support that bsdtar provides to archive additional attributes. So I chose bsdtar so that only 1 tool was required. Unfortunately, whilst PINN can include the extraction part of bsdtar, it cannot support the archiving part. Hence PINN has to use gnu tar for OS backup. This may mean that certain OSes cannot be backed up, but the recent addition of storing ACL info may mitigate this.

closed 2022-03-27 by procount